RasPlex/RasPlex #329: Upgrade drivers of RasPlex/OpenELEC.tv

eplouhinec, 2015-01-03:
Hi!

The USB Wifi Dongle "Edimax AC600 - EW-7811UTC" is not compatible with the latest version of Rasplex (0.4.1).
It seems that the driver RTL8812AU could resolve the problem. This driver has been integrated in openelec in 21 sept. 2014 by these commits : https://github.com/OpenELEC/OpenELEC.tv/commits/openelec-4.2/packages/linux-drivers/RTL8812AU

Is it possible to upgrade RasPlex/OpenELEC.tv to a newer version of OpenELEC? Or just include these drivers?

I tried to apply the fix by myself but with no success. My "./buildman --wip -r -f -i" is working since 2 days without stopping until now !...

dalehamel, 2015-01-06:
These seem hardmless enough. I'll add these patches in the next release.

dalehamel, 2015-01-06:
Ok, I've pulled these drivers in, as well as the drivers for the RTL8192DU
chipset. @eplouhinec would you mind trying out a test build?

dalehamel, 2015-01-06:
@eplouhinec I've added these to the latest build, would you mind trying out an experimental test build to verify support?

dalehamel, 2015-01-06:
@eplouhinec if you have the time, try flashing http://updater.rasplex.com/beta/rasplex-wip.img.gz. It is the current WIP build, it should have support for your wifi card. Note that there is a known bug where local media server discovery doesn't work without myplex, or manual ip configuration and a reboot.

eplouhinec, 2015-01-09:
I've tryed your img and it recognize my wifi card!
Great job! :p

Otherwise, for the build of the project via the vagrant VM, is it normal that it seems download and compile all the world during many hours for the first build ?

dalehamel, 2015-01-09:
@eplouhinec yes, it's built entirely from source. It has to build the operating system and all libraries in the first build. Subsequent builds it will just build plex, or any libraries that changed.

If you're interested in development, we can add you to our beta team if you think you'll have the time. 

I'm closing this for now though, since the issue will be resolved in the next releas.

closed 2015-01-09 by dalehamel

vooja, 2015-01-17:
@dalehamel, @eplouhinec  - could you make this wip build available somewhere? I have the same wifi card and am trying to get it to work. 
I tried building rasplex image from source (using native development environment instructions - https://github.com/RasPlex/RasPlex/wiki/Native-development-environment-setup), but ran in to a bunch of issues:
http://sources.rasplex.com/devel/misc-firmware-0.0.6.tar.xz not found (updated that to 0.0.12), then a few others. 
Eventually got stuck at:
RasPlex/OpenELEC.tv/build.rasplex-RPi.arm-devel-release/plexht-wip/plex/Resources/system/*: No such file or directory

dalehamel, 2015-01-17:
these drivers are included in 0.5.0

vooja, 2015-01-17:
oh! i started by trying 0.5.0 first, do not see wireless networks option in system settings. sorry, if my questions sound noob-ish, that is because i am a noob to rasplex world

vooja, 2015-01-17:
also, got latest openelec build, and that was able to identify my wifi card, and connect to network fine. any suggestions on how to debug this?

dalehamel, 2015-01-17:
what is the chipset on your card

vooja, 2015-01-17:
this is the card i am using - https://wikidevi.com/wiki/Edimax_EW-7811UTC
on my ubuntu box, i was able to get it to work using this - https://github.com/gnab/rtl8812au

thanks for the super fast responses :-)

dalehamel, 2015-01-17:
the rtl8812au driver should be included in 0.5.0, if it's not that's a bug.
you may need to try ssh'ing in and modprobing it manually. do you see it
under /lib/modules anywhere?

vooja, 2015-01-17:
rasplex Version: 0.5.0
rasplex git: 3251ac74fad91a86af2f3ea56ac661ea60038827

RasPlex:~ # lsmod
Module                  Size  Used by
bluetooth             215643  4 
iptable_filter           967  0 
ip_tables               9755  1 iptable_filter
lirc_rpi                6012  3 
lirc_dev                7885  1 lirc_rpi

RasPlex:~ # modprobe 8812au

RasPlex:~ # lsmod
Module                  Size  Used by
8812au                967960  0 
cfg80211              313952  1 8812au
bluetooth             215643  4 
iptable_filter           967  0 
ip_tables               9755  1 iptable_filter
lirc_rpi                6012  3 
lirc_dev                7885  1 lirc_rpi

RasPlex:~ # ls /lib/modules/3.10.20/
RTL8188CU            modules.alias        modules.devname      modules.pcimap       modules.usbmap
RTL8188DU            modules.alias.bin    modules.ieee1394map  modules.seriomap
RTL8188EU            modules.ccwmap       modules.inputmap     modules.softdep
RTL8812AU            modules.dep          modules.isapnpmap    modules.symbols
kernel               modules.dep.bin      modules.ofmap        modules.symbols.bin

vooja, 2015-01-17:
After I modprobe it manually, I still do not see the interface in system prefs (in the GUI). Is there a way to get the module to load on boot?

dalehamel, 2015-01-17:
after probing, does anything new show up in /sys/class/net?

vooja, 2015-01-18:
/sys/class/net does not get anything new after the modprobe (only eth0 and lo0 before and after)

vooja, 2015-01-18:
lsusb shows that the device is indeed seen by the OS, any clues on what is making it not create the wlan0 device?

RasPlex:~ # lsusb
...
Bus 001 Device 004: ID 7392:a812 Edimax Technology Co., Ltd

osvaldolove, 2015-02-08:
Hi, just tried the above as I have the same WiFi dongle and it isn't working using 0.5.0.
I created /storage/.config/autostart.sh with the following to force load the module on boot:

#!/bin/sh
(modprobe 8812au;)&

No wlan shows up in /sys/class/net or through the Plex GUI though. Any ideas?
Thanks...

dalehamel, 2015-02-08:
Hi,

This issue should be resolved in the next release (0.5.1) which will be
coming out in the next couple of days. There was an issue with the au
driver that was recently patched.

-Dale

osvaldolove, 2015-02-08:
Great stuff... thanks a million!