RasPlex/RasPlex #330: Rasplex font bugs, very slow and some crashes

oeppesd, 2015-01-08:
Hey,
I have a problem pretty similar to this one: https://github.com/RasPlex/RasPlex/issues/294
I'm using rasPlex 0.4.1 on my pi Model B. And sometimes the font gets all buggy. The user interface itself seems kind of unstable, sometimes it doesn't show the Coverarts of Movies, and rasPlex crashes a lot, so it is unusable for me at the moment. Can anyone help me with this?

NedtheNerd, 2015-01-08:
Run free -m on the command line and post back the output, this is only a problem on Model A's or 256Mb Model B's, neither are supported by Rasplex.

Regards

dalehamel, 2015-01-09:
Also, what class of SD card are you using

NedtheNerd, 2015-01-09:
Yep, it's a 256mb Model B, sorry, nothing we can do.

If you want to continue to use Rasplex you will have to source a new Raspberry Pi.

SD Class is printed on the card itself, look for the number 10, 6 or nothing at all, only Class 10 are supported.

Closing the issue.

closed 2015-01-09 by NedtheNerd