RasPlex/RasPlex #331: HDMI CEC lost

yannlet, 2015-01-11:
This bug occurs in this specific sequence:
- TV is switched to a tv channel
- Rasplex runs in the background and after x minutes, screensaver starts
- If then I want to switch back to Rasplex, the HDMI CEC signal is lost, and I can't use the TV remote control to select the user

Temporary solution : disable screensaver and then I can switch between TV channels and Rasplex without issue

Raspberry PI 512Mb

dalehamel, 2015-01-11:
Thanks for this report and work-around.

Can you tell us what model of TV you have?

NedtheNerd, 2015-01-11:
I have never had this issue with two different Sony Bravia TVs, have you checked you CEC settings on both the TV and Rasplex.

yannlet, 2015-01-12:
My TV is a samsung PS65C7700, and yes I checked the CEC settings on both sides.
I have been using plex since the beginning, and never had this issue and today if the screensaver is disabled, no issue as well.
I don't know if it makes sense but it's like if the "switch user" screen doesn't trigger CEC anymore

yannlet, 2015-01-12:
Just checked with another samsung TV UE32C4000 .... and exactly same problem

dalehamel, 2015-01-13:
This is likely a new big introduced by plex home

NedtheNerd, 2015-01-13:
I think this was using 0.4.1, @yannlet can you confirm which Rssplex version you are using.

yannlet, 2015-01-13:
0.5.0
I am referring to the "switch user" screen introduced with Plex home to select users ...... didn't exist before in 0.4.1 :-)
@dalehamel you are probably right - I tried on 2 different TVs with 2 different raspberry pis: same result

NedtheNerd, 2015-01-13:
Have you checked the 'Automatic login' in Preferences, System, Plex?

NedtheNerd, 2015-01-13:
See this post in Plex HT Releases https://forums.plex.tv/index.php/topic/56516-plex-home-theater/?p=793052

yannlet, 2015-01-13:
Yes 'automatic login' is another workaround (like disabling screensaver), but I would like to avoid it - I would like to keep the switch user screen after an inactive period because my son is 11 years old and I want him to use his Plex account (otherwise I have to think to disconnect from my Plex account every time I leave the living room)

NedtheNerd, 2015-01-13:
Just fired up my test Plex Home PMS server to investigate, will report back.

NedtheNerd, 2015-01-13:
Have now been testing this.

On my Sony Bravia TV, leave Rasplex on user with a PIN, switch TV input to TV, watch TV for 10 minutes then switch TV input to Rasplex, touch any key, select user dialogue displayed, press OK, enter PIN, Home Screen displayed.  Leaving Rasplex on any other user without a PIN and returning from TV input to Rasplex input, no select user dialogue displayed.  Results as expected - no 'bug'.

Now this means that the CEC on Sony TV's don't manifest the issue you have, and it is a well known fact that not all CEC implementations are the same for different manufacturers.

For info my CEC settings are as follows:

Enabled - checked
Make XBMC the active source when starting - unchecked
Devices to power on when starting XBMC - None
Devices to power off when stopping XBMC - None
Send 'inactive source' command when stopping XBMC - unchecked
Put devices in standby mode when activating screensaver - unchecked
When TV is switched off - ignore
Put devices in standby mode when putting the PC in standby - unchecked
Use the TVs language setting - checked
Pause playback when switching to another source - unchecked
Physical address (overrules HDMI port)
COM port(leave empty unless needed) - '-'

You might want to try these.

Regards

yannlet, 2015-01-13:
I will check my settings tonight and let you know. As you said, unfortunately, all CEC implementations are not the same for different manufacturers, maybe it is specific to Samsung TVs.
Thank you for your time !

yannlet, 2015-01-13:
I just checked with those settings ... no change, still same issue .... unfortunately

DJFliX, 2015-02-27:
I have the same issue. I recently purchased a Raspberry Pi 2 B and installed RasPlex 0.5.1 using the installer for Mac. On my Raspberry Pi B (no +) I did not encounter this bug. I'll check whether that was version 0.5.1 as well. I'm using a Samsung UE40D5000.

DJFliX, 2015-03-13:
I just verified that the version of RasPlex that I ran on my Pi Model B is 0.5.1. So I guess I'll try to pull the configs from the old SD card and compare them to the configs on the Pi 2 Model B. 

I must note that the problem is somewhat circumvented by remotely starting to play something (using the Plex Web interface) and turn then turning on the TV.

Reilos, 2015-05-13:
On my Samsung TV model UE48HU7500, Raspberry Pi 2, RasPles v 0.6.0. Same issue. If the screensaver turns on for a pin protected user while TV is off or just watching regular TV, the select user screen does not respond anymore.

NedtheNerd, 2015-05-13:
Next release will have an updated libCEC, hopefully this will fix the CEC issues in 0.6.0.

norbusan, 2015-06-12:
Oh, so that could be related to my CEC issues (#378 which you closed recently). I found out that it is a very unstable procedure to get CEC working. I need to do the following steps to get CEC working in most cases:

* unplug all devices
* unplug TV from electricity
* start TV, make sure you are on a TV channel
* go into BraviaLink CEC list of devices - there is nothing
* while there, connect the Rasplex HDMI cable (rasplex still turned off!)
* then connect the Rasplex power source
* after a short time the screen switches to RasPlex booting

After this procedure I normally have CEC running on my Sony Bravia/Raspberry Pi2/Rasplex 0.6.0 combination.

Hope that the next update brings some better ways!

Thanks and all the best

Norbert

Kwiboo, 2015-06-12:
@norbusan Check out http://kodi.wiki/view/Raspberry_Pi_FAQ#TV_is_not_detected_unless_powered_on_first, it could possibly help with your issue.
Next release will also have lots of other CEC fixes.

norbusan, 2015-06-12:
Hey, thanks a lot @Kwiboo that was a great tip. I have tried two start-up combinations that didn't work before and this time they worked. I will play around a bit more and report back, but it is good to know that there is this workaround!

Thanks again

Norbert

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

Reilos, 2015-07-19:
Works like a charm now! I'll keep an eye on it.

norbusan, 2015-08-18:
Looks better, thanks a lot.

NedtheNerd, 2015-09-11:
Issue appears to be fixed from comments, closing issue.

closed 2015-09-11 by NedtheNerd

tangb, 2017-01-12:
Hi i'm encountering the same issue in my sony Bravia (from 2009)
Disabling screensaver doesn't resolve the problem.
I think the problem appears after I play a media. If I reboot and do nothing, the rasplex is still available on my tv source menu.
The raspi is running, the activity led is still blinking. The only way to turn it hdmi on is to launch a media playback using my phone.

I'm running latest rasplex release (1.7.1) from a raspi3 and class 10 sdcard. It's a fresh 1.7.1 rasplex install, not an update from stable release (1.6.2)

Thank you for your help.