RasPlex/RasPlex #332: [Bug] Linux installer --> black screen on rPi

SuperManitu, 2015-01-14:
Today I tried to set up my Rasplex, but the Raspberry didn't show any GUI, but SSH worked.
I solved the issue with setting the "hdmi_drive=1" to "hdmi_driver=1"
I think there is a spelling mistake in that config file after installation.

NedtheNerd, 2015-01-14:
Don't think so, config.txt only contains hdmi_drive=2 by default, suggest you spelt it wrong when you edited the file.

Closing issue.

Regards

SuperManitu, 2015-01-14:
I mean drive should be driver, shouldn't it?

NedtheNerd, 2015-01-14:
No, it's hdmi_drive, see here for all config options, most of which are not normally required, only those in the deafault config.txt.

This is not a bug, leave closed please.

closed 2015-01-14 by NedtheNerd

g1t-dlanor, 2015-01-14:
@SuperManitu:
To further clarify the "hdmi_drive" usage:
The default value 2 as used in the original RasPlex image is the proper one for normal HDMI TV sets.
DVI monitors use different HDCP protocol variants, and for them you need "hdmi_drive=1".
Your mention of having an original value of "hdmi_drive=1" implies that you either edited it so yourself, or got the config.txt file from another source than the original RasPlex release image.
That could make RasPlex fail to display properly on a normal HDMI TV set.
And your changing the variable name to "hdmi_driver" would in fact make that setting be ignored by the system routines, making them fall back on an internal default, which is why it worked for you.
But that variable name is still wrong, the release default is still right, and the issue remains closed.