RasPlex/RasPlex #335: Screen gets resized after TV is turned off.

jmiller-sc, 2015-01-31:
Every time I turn the tv back on to use Rasplex the screen is all squished to a small resolution. I then have to go through settings screens to enlarge it again. I have read something about taping up pins on HDMI cables to fix this. However, that seems like an extreme solution when there are many pi builds that don't seem to have trouble maintaining screen size after the TV is turned off and on.

g1t-dlanor, 2015-01-31:
I see two possible reasons for this behavior:

[u]Case 1: Initial HDMI handshaking failure[/u]
If the RPi is powered from a USB connector of the TV set, then power may be applied to it as soon as you turn the TV on, even though the TV software is not yet ready to interact properly with peripherals (it often takes several seconds of initialization before this works) and this may cause the HDMI handshaking to find a compatible resolution to be performed by RasPlex before the TV is ready to respond. Such failure frequently leads to default choice of the most universally compatible resolution of 480p (nearly all TV sets and PC monitors can support this). This can also happen sometimes when using an HDMI hub, which doesn't auto-switch as wanted (or fast enough) to connect RasPlex to TV during the handshaking moments.

Workaround:
Instead of correcting the resolution through the Preferences submenus you could just use the sidemenu command "Restart". This will cause the RPi to reboot with the TV already active and ready for handshaking in this new reboot. This should cause the highest compatible resolution to be used.

Real solution:
Edit the "config.txt" file on the SD card to make it enforce a specific resolution regardless of the handshaking with the TV. This will work, but has the drawback that if you connect the RPi to some other TV which doesn't support that resolution you'll only get a black screen...
Suitable entries to add near bottom of "config.txt" would be:
For 720p 60Hz use "hdmi_mode=4"
For 1080p 60Hz use "hdmi_mode=16"
This assumes that you are using a normal TV set, as implied by your mention of it as a TV.

[u]Case 2: HDCP/CEC interactions apart from initial handshaking[/u]
Such interactions can mess up operations in countless different ways, due to manufacturer differencies in how the protocols are implemented.
What you then need to do is to configure the HDCP/CEC interactions between RasPlex and your TV set. Some TV manufacturers have their own names for these protocols (Viera etc), but basically any protocol allowing equipment to interact through an HDMI connection (apart from just transferring the media signals) is always related to HDCP/CEC, which are manufacturer-independent definitions. Even so, the protocols used by manufacturers differ, and are sometimes incompatible. You should be able to turn such interactions on/off from the TV menus, and you can also reconfigure the CEC adapter settings of RasPlex.

To do this, go to the submenu 'Preferences > System > "Input devices" > Peripherals' where you will find an entry named "CEC Adapter" which can be opened to reveal a list of settings controlling how and if RasPlex will interact (or at least try to do so) with a TV set. Using those settings should allow you to eliminate unwanted interactions.

There are also some "config.txt" settings that you can add to affect CEC usage.
To disable CEC for initial setup use "hdmi_ignore_cec_init=1"
To make the RPi act as if the TV has no CEC at all use "hdmi_ignore_cec=1"

Using some of the above methods you should be able to eliminate the problem.
If/when you do fix the problem, please come back here and close this issue.
I'll leave it open for now, just in case the problem is more complex than I think it is.

jmiller-sc, 2015-01-31:
Thank you. I really appreciate your quick and detailed response.

I will add clarifications on each point before trying some things out and following up here:

* The RPi is not connected to the TV for power.
* An HDMI hub is not used in the way that I am thinking of a "hub". However, it is plugged into a Denon receiver which essentially probably isn't any different than a hub.
* There are no plans to plug it into another TV. And the size is pretty standard for HD (1920x1080). So, hard-coding it that way would be a good solution for me, assuming that it works. I will try after testing some other options.
* I suspect that CEC is the problem. In addition to this resizing issue, we also have an issue where when I turn the TV off and then on, the receiver automatically switches to the RPi. OR, it could be that CEC has other devices go to sleep and it switches to the RPi because it's the only device returning a live ping?
* I have already attempted to fix that CEC issue via Rasplex by disabling CEC features. But, maybe I hadn't yet completely disabled it but just some features. (I do use CEC to have the TV send volume change requests to the receiver which is why I haven't turned it off on the TV side).

Will poke around with all of these things and follow up to let you know how it goes.

dalehamel, 2015-02-13:
Try hdmi_drive=1

NedtheNerd, 2015-05-22:
Any update on your issue?

jmiller-sc, 2015-05-26:
I believe that a Rasplex update fixed this issue.

closed 2015-05-26 by jmiller-sc