RasPlex/RasPlex #336: Raspberry Pi 2 Support

microadam, 2015-02-02:
http://www.raspberrypi.org/raspberry-pi-2-on-sale/

Will you guys support the new Rasperry Pi 2 anytime soon? I believe a software upgrade is required for support.

Thanks

dalehamel, 2015-02-02:
Yes, but unfortunately we don't have any hardware yet. I ordered some today, but the supplier is out of stock. Without test hardware, there's no way we can put a release out. The estimated ship date is Feb 16th.

microadam, 2015-02-02:
Makes sense! Glad to know it is on the horizon though!

If it would help at all, I should have mine tomorrow. Happy to help test things out?

dalehamel, 2015-02-02:
There is quite of lot of iterating that happens behind the scenes. Without
an actual unit to test on directly progress will be very slow.

microadam, 2015-02-02:
Figured that would be the case! Look forward to hearing of the progress once you get your test units!

dalehamel, 2015-02-02:
I'm really pumped for this, it's putting quite a lot of juice into the
little raspberry. I'm concerned that it might take a while to get working
though, as the XBMC codebase we use is quite old, and the Helix merge is
still in progress. We're also using a very old OpenELEC codebase, because
of compatibility reasons with Frodo.

Then again, it may be as easy as just changing some compile flags and
dropping the new drivers in, which would be a few days work. It's hard to
say how hard this is. Nevertheless, we won't stop until it's done.

dalehamel, 2015-02-13:
The latest release (0.5.1) supports RPi2

closed 2015-02-13 by dalehamel