RasPlex/RasPlex #343: ensure hifiberry is supported

dalehamel, 2015-02-13:
Lots of people want this, let's make sure we do in fact support it properly

microwavesafe, 2015-03-21:
Can I ask whether this will be fixed in the next release, which I guess is 0.6.0.

BradMeier, 2015-03-21:
No, it isn't planned for 0.6.0, hopefully the next release after that.

microwavesafe, 2015-03-21:
Is 0.6.0 in feature freeze or is this something I can help with?

NedtheNerd, 2015-03-21:
Yes it's frozen, hoping to get this out soon.

microwavesafe, 2015-06-20:
I wondered if there were any further updates on this issue? Do you think there will be a fix in 0.6.1?

NedtheNerd, 2015-06-20:
HiFiBerry is supported in next release for music playback only

microwavesafe, 2015-06-20:
Interesting, is different software used for video and music playback? I'm guessing omxplayer is used for video so is it a limitation with this?

microwavesafe, 2015-06-20:
Does openelec 5 use ALSA as the back end?

NedtheNerd, 2015-11-25:
Need togged HiFiBerry (and USB DAC's) working for video playback.

NedtheNerd, 2016-03-30:
Fixed in next release, closing issue.

closed 2016-03-30 by NedtheNerd