RasPlex/RasPlex #345: Fix on deck/fanout issue

dalehamel, 2015-02-13:
Sometimes fanouts don't show up, it seems to happen more often when signed into MyPlex.

Reports have said this helps fix it:

Settings -> Network -> Advanced Network Settings -> Wait for network before starting XBMC, this seems to fix that on-deck issue.

dalehamel, 2015-02-13:
https://github.com/RasPlex/service.openelec.settings/blob/rasplex/resources/lib/modules/connman.py#L2573

Might be fixed by new openelec settings @kwiboo

dalehamel, 2015-02-13:
Should also be fixed in plex code: 

xbmc/network/Network.cpp
171:bool CNetwork::IsAvailable(bool wait /*= false*/)

dalehamel, 2015-02-15:
Adding the IsAvailable call before plexapplication is started seems to resolve this

NedtheNerd, 2015-02-28:
Appears to be fixed in next release.

closed 2015-04-21 by NedtheNerd