RasPlex/RasPlex #350: Hyperion with stk1160 driver still not working

posthok, 2015-02-13:
Hyperion with stk1160 driver still not working in version 5.

dalehamel, 2015-02-13:
@posthok where can we get this hardware so that we can verify it?

It seems like this is not yet working on OpenELEC either

https://github.com/OpenELEC/OpenELEC.tv/issues/3503

posthok, 2015-02-14:
I've got it working on the following image.
OpenELEC 5.0 with stk1160-compatible kernel B+ compatible
Link: http://img.lightberry.eu/download/OpenELEC-5.0.RPi.arm-Bplus_stk1160.zip

dalehamel, 2015-02-14:
@posthok do you know what patches were needed to do this?

posthok, 2015-02-14:
Sorry thats beyond my knowledge. Not my skills.
Maybe you can contact the owner of the lightberry site who has the images of openelec with the drivers. The email adress is lightberry@lightberry.eu and his name is Jacek. Maybe he can support you,

dalehamel, 2015-02-24:
@posthok I've added the necessary patch, and have asked OpenELEC to also apply it. See https://github.com/OpenELEC/OpenELEC.tv/pull/3948

I purchased a lightberry and can confirm that it correctly grabs the frames. This will be available in the next rasplex release regardless of if it is merged into openelec.

closed 2015-02-24 by dalehamel

posthok, 2015-02-24:
Thanx Dale you're the best !!

mmodler, 2016-08-11:
@dalehamel

Just to summarize: OpenElec still hasn't applied the patch. But its in RasPlex? So i might get rid of the issue switching to a rasplex image?

Thanks!