RasPlex/RasPlex #353: GPIO IR receiver not working in 0.5.1?

jishi, 2015-02-13:
I have a pi B that has been running rasplex for some time now, with a GPIO IR receiver. Today I upgraded to 0.5.1 and now I notice the strange behavior that I do get a /var/run/lirc/lircd socket, but in dmesg I see this:

TV:~/.config # dmesg |grep lirc
[    5.594741] lirc_dev: IR Remote Control driver registered, major 248
[    5.623922] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[    5.629108] lirc_rpi: gpio chip not found!

using "irw /var/run/lirc/lircd" I can't see any output at all when I press my remote, which I do think should be the case. Is there something that has been removed from the kernel perhaps? Is there a simple way to downgrade without reformatting the SD (so I can keep the config) to double check my config with 0.5.0 again?

NedtheNerd, 2015-02-13:
Yes, the kernel was changed by raspberrypi.org for all models of Raspberry Pi and we are now using the latest kernel

You may need to do some work yourself to get this working again if you are using GPIO rather than USB Lirc devices.  See http://www.raspberrypi.org/documentation/configuration/device-tree.md.

This is not specific to Rasplex none of the overlays are loaded by default in Raspbian.

NedtheNerd, 2015-02-13:
This is what I see:

rasplex Version: 0.5.1
rasplex git: 55113f6ddae290c431fdf96171c28c1778fbe1ff
RasPlex:~ #  dmesg |grep lirc
[   15.990445] lirc_dev: IR Remote Control driver registered, major 248
[   16.023667] lirc_rpi: module is from the staging directory, the quality is unknown, you have been warned.
[   16.940624] lirc_rpi: auto-detected active low receiver on GPIO pin 18
[   16.941037] lirc_rpi lirc_rpi: lirc_dev: driver lirc_rpi registered at minor = 0
[   16.941058] lirc_rpi: driver registered!
[   17.844065] input: lircd as /devices/virtual/input/input0
RasPlex:~ #

jishi, 2015-02-13:
Adding 
device_tree_overlay=lirc-rpi

to config.txt enabled it again. Maybe that was added to the image-release but I seem to have my old config.txt since before upgrade, that would explain why it was loaded for you?

Anyway, thanks for the point in the right direction!

closed 2015-02-13 by jishi