RasPlex/RasPlex #355: Adding the ability for USB DAC enhancement

travis-pettry, 2015-02-19:
The audio quality and volume on the Raspberry Pi are low. There are currently no solutions (other than Hifi berry) for making the audio cleaner or louder. With platforms like XBMC that have the USB DAC implemented would it be possible to do the same for rasplex?

dalehamel, 2015-02-19:
point us at a USB DAC solution, and if you're so inclined you can donate
one to the project for testing. Come from XBMC would be helpful too.

NedtheNerd, 2015-02-20:
There are far too many different USB DAC's, most of those on the market are meant for PC's and are mostly plug and play, however, Rasplex is a Linux based system and many 'plug and play' devices simply don't work.  I assume that you have done the usual Internet searches and have drawn a blank, it's likely that we would end up in the same position.

The sound quality from the audio jack is abysmal, the B+ is slightly better, if you cannot output sound via HDMI then I would suggest looking at the HiFiBerry, the cheapest, reliable DAC for the Pi.  I have tried some of the cheap USB DAC's in the past on PC's and found them not much better than the output from onboard sound cards.

mdvorak, 2015-05-04:
OpenElec 5.x works with USB sound card almost out-of-the-box (had to disable GPU Acceleration for audio). Simply boot with usb card plugged in, and select it in the menu.
Since RasPlex 0.6.x is based on OE 5, it should be just matter of merging the functionality, if I'm not wrong.
PS: I know its not "just merge", but it should be possible.

TheOriginalCoda, 2015-06-20:
I'm currently testing many different permutations of XBMC forks on the Pi and Pi2 to see what will work best for me. I've got a USB soundcard (creative X-FI) to work on OpenElec 5.08/Kodi/PleXBMC. The device just showed up as available in the audio settings, and once I switched to that instead of HDMI I get 5.1 AC3/AAC/DTS over TOSLINK. If RasPlex is using OpenElec, why isn't it also working in RasPlex?

NedtheNerd, 2015-06-20:
Because current release of Rasplex does not use OE 5.0.8, next release will, so it may work, or not!

fabienheureux, 2015-09-23:
Hi @NedtheNerd, about your last comment, does Rasplex now have USB DAC implemented ?

NedtheNerd, 2015-11-09:
USB DACS should now work with 1.0.0, please check and advise.

1Map, 2015-11-24:
Still not working in 1.0.0

I have a Raspberry Pi 2 running latest Rasplex 1.0.0, with Behringer UCA222 USB DAC
http://www.amazon.com/Behringer-UCA222-BEHRINGER-U-CONTROL/dp/B0023BYDHK

Still I cannot see an option for USB DAC.

Please advice.

g1t-dlanor, 2015-11-25:
@1Map :
I would not expect every USB DAC/Soundcard to work even with the new RasPlex version.
We are  in this respect limited to what's supported by the OpenELEC version RasPlex is based on.

Your DAC appears to be a rather complex model, as the ad describes it as being intended for complex processing of both input and output audio. This could mean that it requires fairly complex drivers, and while these may be bundled with both Windows and OSX, that doesn't mean that any are available for OpenELEC on the ARM CPU of RPi. (Though I don't know whether it is or not...)

Have you tested that your DAC device works with Kodi in OpenELEC 5.0.8 on the RPi ?

If it does, then it should be possible to make it work in RasPlex as well.
But if it doesn't, then I guess OpenELEC simply lacks a compatible driver.

1Map, 2015-11-25:
It works 100% in Kodi / OSMC / OpenELEC.  Problem is I do not know how to setup it in RasPlex?

NedtheNerd, 2015-11-25:
Have you looked in Preferences->System->Audio Output->Audio output device (towards the bottom of the options) and select your device there?

1Map, 2015-11-25:
Found it.  It was way at the bottom on my small touch screen.  I was looking at the first option Audio Output, which just gave me HDMI, Coaxal, etc.  Bit confusing.  Anyway it works now 100%.

NedtheNerd, 2015-11-25:
Have now tested 3 different USB DAC's, all working (music only), confirmed by user also, closing issue.

closed 2015-11-25 by NedtheNerd

firsttris, 2015-12-06:
Hello, my usb-dac (nubert no-pro) is not working with rasplex 1.0.1.

i have the same settings like in openelec. (working in openelec)

audio device: AL:SA USB AUDIO DAC, SPDIF
passthrough: ALDA USB AUDIO DAC, S/PDIF

no matter witch setting i choose it plays from hdmi.