RasPlex/RasPlex #356: PHT freeze when clicking next in "play random all"

Eloo, 2015-02-20:
Hey guys,
i recently updated to RasPlex 0.5.1 and figured out an issue when playing random episodes from a TV show.
Evertime when i want to skip a episode by clicking "next" on my remote PHT is freezing. In the bottom right corner there appears a "please wait" notification but PHT seems to be dead. 

I don't know if the problem is new in the 0.5.1 version or its actually a RasPlex problem. But maybe someone other got the error too.

Infos:
Raspberry1 B Model 512MB
System is still reachable using SSH
CPU Load ~ 20% (mostly by PHT)
Killing PHT fixes the problem
PlexServer is available and reachable

If you need more information let me know

Regards Eloo

dalehamel, 2015-02-21:
This will likely be fixed in the next release

Eloo, 2015-02-21:
Very cool.
And i just figured out that PHT is not freezing when I'm using the on-screen option for "next" (from the small menu on bottom of the screen)

eraserhead666, 2015-04-06:
I experience the same when selecting next episode using the on screen menu. : 
[image]

derekcentrico, 2015-05-08:
Glad to see this is reported.  I'm in the same boat with freezing.

NedtheNerd, 2015-05-21:
Not sure what you mean by 'Play Random All', can you describe the steps to replicate this.  Thanks.

derekcentrico, 2015-05-21:
For me, I will select the shuffle feature say for all seasons of my favorite show SouthPark.  If I try to skip to the next episode, I will likely get the freeze of "Please wait" until I power cycle the rPi2.  But if I just jump forward a few times to the end of the current playing show, Rasplex will play the next episode fine.

NedtheNerd, 2015-05-21:
This issue is fixed in next release.

Will leave open for now, when next release is out can you check and close if appropriate.

derekcentrico, 2015-06-07:
Ned, I can test now if you want.

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

derekcentrico, 2015-07-20:
I haven't experienced it yet.  Close it out and we reopen if needed?

NedtheNerd, 2015-07-20:
Closed at request of OP.

closed 2015-07-20 by NedtheNerd