RasPlex/RasPlex #358: Bluetooth input/output error when connecting to anything

biltongza, 2015-02-27:
I have a Kensington Bluetooth 4.0 adapter plugged into my RPi 2 running rasplex 0.5.1 and rasplex picks things up no problem but can't seem to connect to anything with it, just giving a generic input/output error.

If you try connect it will initiate a pairing request just fine, with the correct code appearing on both devices, but after that is when I get the I/o error.

I'm not entirely sure of the exact model number or what chip set the dongle actually uses but if someone can tell me what to look for I am more than happy to give more details if I can find them.

Velociraptor85, 2015-03-22:
I also have problems with my Bluetooth dongle and a Bluetooth keyboard, but i found that it works with the current OpenELEC so my question is more like which version of OpenELEC 0.5.1 uses.

Also the Problem is there for the Stable tree.

Velociraptor85, 2015-05-01:
I hoped that this would be resolved with a newer 0.6.0 but my bluetooth keyboard still does not work

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

biltongza, 2015-07-19:
Unfortunately it seems that now my wireless adaptor isn't working in 0.7.0,
nor is my Bluetooth dongle. I didn't get a chance to look at it properly
today, I will try again tomorrow.

biltongza, 2015-07-20:
Ok, so wireless networks are disabled by default and I completely missed the setting to enable it last night. Bluetooth dongle is working too, but I still can't connect to my phone (input/output error, file or directory not found is the toast I get) and now there are zero messages in the log file anywhere that I can see that could be related to bluetooth. Log is [here](http://pastebin.com/BAte7f3h) in case you can fathom something out of it.

Good news is that traceback in dmesg is no longer there. [dmesg.log](http://pastebin.com/A7XMWrGe)

Velociraptor85, 2015-07-25:
So I upgraded my RPi1 to 0.7.0 and the Error that regarding the connection is gone it connects and it shows up.

BUT

It does not use the Device doesnt matter which keys I use or stroke.
As @biltongza said there are no error messages and bluetoothctl and btmon show up the device and the Keystrikes so it has to do something with piping in the device to the PHT or something.

The nice thing is that even after an PI reboot the device is remembered and connected to.

Furthermore the bluez.conf is there so thats not a problem.

adrianmihalko, 2015-07-27:
Same problem here, I am connected to my Apple keyboard but keypress is not recognised.

NedtheNerd, 2015-08-23:
Fixed in next release.

NedtheNerd, 2015-09-08:
@biltongza

Can you check in 0.7.1 and advise

biltongza, 2015-09-08:
Nope, still got the same error. "Bluetooth error No such file or directory".

NedtheNerd, 2015-09-08:
What are you trying to connect to? Bluetooth keyboards connect without issues and survive restart.

Kwiboo, 2015-09-08:
@biltongza: Please `Delete` any existing connection and use `Pair` to connect your device again. Host initiated connect do not always work and will sometime return as input/output or no such file or directory error.
Check the output of `journalctl --no-pager --all` for any problems relating to `bluetoothd`.
Also check for `RasPlex Addon` lines in `plexhometheater.log`.

LegeDoos, 2015-09-13:
Same issue here on latest rasplex 0.7.1 when connecting to Bluetooth device: Bluetooth error, no such file or directory. Even after deleting and reconnecting.
I try to connect my Nokia Lumia 920. Want to stream music. Don't even know if that is supported :)

NedtheNerd, 2015-09-14:
Bluetooth support only allows keyboards, no support for steaming, file transfer, Bluetooth speakers etc.

LegeDoos, 2015-09-14:
Thanks for the info. Then this bug isn't that hot for my anymore :)

NedtheNerd, 2015-11-09:
Closing issue, only keyboards are supported under bluetooth.

closed 2015-11-09 by NedtheNerd