RasPlex/RasPlex #359: Miracast support

biltongza, 2015-02-27:
I would like to see miracast support included in rasplex if at all possible. There is a github repo here https://github.com/codemonkeyricky/piracast that got it working on raspbian but I'm not sure exactly what is required to get it working on rasplex. I have a RPi 2 with the WiFi adaptor the project requires and I'm more than happy to try things out if needed.

dalehamel, 2015-02-27:
We could look into it, but it looks like this is some microsoft thing? Might be hard for us to test.

NedtheNerd, 2015-02-27:
For information.

Miracast is a certification program of the Wi-Fi Alliance based on their Wifi-Display specification. It defines a protocol to connect an external monitor or TV to your device, and therefore can roughly be described as "HDMI over Wifi", replacing the cable from the computer to the display.

It is peer-to-peer, and wireless, using a Wi-Fi Direct connection. It allows sending up to 1080p HD video and 5.1 surround sound (AAC and AC3 are optional codecs, mandated codec is linear pulse-code modulation — 16 bits 48 kHz 2 channels). The connection is created via WPS and therefore is secured with WPA2. IPv4 is used on the internet layer. On the transport layer, TCP or UDP are used. On the application layer, the stream is initiated and controlled via RTSP, RTP for the data transfer.

Adapters are available that plug into HDMI or USB ports that enable Non-Miracast devices to connect via Miracast. The developer responsible for the implementation of Miracast in Linux does not recommend the standard as the technology is "horrible".

g1t-dlanor, 2015-02-27:
'Horrible' or not, I don't see how this can be made part of the RasPlex project as such, since the RasPlex project is focused on porting the "Plex Home Theater" application to the platform of OpenELEC on Raspberry Pi, not to implement new low-level interface drivers.
If this form of WiFi-based video output is implemented at all it should be done by additional device driver support added to OpenELEC, after which we should be able to use that device driver from RasPlex-PHT just like we use the normal HDMI drivers provided by the OpenELEC OS. The changes to RasPlex code would then mainly consist of some routines to enable this output mode, switching which drivers are used to handle the HDMI data, with low-level drivers of OpenELEC doing the real work. But to implement such Miracast drivers independently of OpenELEC would in my opinion not fit in with the RasPlex project as it exists today.
So as I see it such a feature request should be made for OpenELEC, not RasPlex.

biltongza, 2015-02-27:
You make a good point. I shall take this to openelec then. :)

NedtheNerd, 2015-02-28:
Closing due to OP's last comment.

closed 2015-02-28 by NedtheNerd

RichardEb, 2015-09-08:
I'm also interested in this feature. Do you have any new infos?

NedtheNerd, 2015-09-08:
No, issue is closed.

Regards

RichardEb, 2015-09-09:
You said that it would be no problem if OpenELEC implement it. Does OpenELEC already implemented it? Does somebody ask them to do that?

RichardEb, 2015-09-09:
https://github.com/OpenELEC/OpenELEC.tv/issues/4306

RichardEb, 2015-09-09:
You should switch to a different base then Openelec. https://github.com/OpenELEC/OpenELEC.tv/issues/4306
Openelec don't seem to be nice base.

fritsch, 2015-09-09:
Yeah and you @RichardEb should start to read and think about open software that makes your pi to a media center. https://github.com/OpenELEC/OpenELEC.tv/issues/4307 as a reference to congratulate future OpenSource projects to have you as users.

RichardEb, 2015-09-09:
i will buy a good planned and developed FireTv instead of your  fiddle  project that no one needs.

dalehamel, 2015-09-10:
This conversation has been locked due to unproductive discussion.