RasPlex/RasPlex #360: Not turning Rasplex on after switching TV on with remote

fxxer, 2015-03-05:
I have a problem to start my Rasplex, after I turned off the TV. Raspberry is suspended and when I switch TV on using remote, Raspberry remains suspended (red light is on) . I am using it with default parameters after fresh install.

Also smaller bug, but still annoying. The Rasplex is detecting TV language wrong. My TV has Slovak language and I am getting Slovenia. Maybe some typo in country shortcodes (Slovak is sk and Slovenia is sl)?

TV: Philips 55PFS6609
Raspberry 2
Rasplex 0.5.1

fxxer, 2015-03-05:
One more thing .. when Rasplex is on controlling it using TV's remote work fine, even after switch the source. Only the starting Rasplex after it was suspended seems to be problem.

NedtheNerd, 2015-03-17:
You need to check what setting you have for 'When the TV is switched off' in Preferences -> System -> Input Devices, I suspect that it is currently set at 'Shutdown', change this to 'Ignore'.

fxxer, 2015-03-17:
It was on Suspend, but I needed to use new SD card and now it's working file. Also now I am using wifi USB and it is working correct. Right now my problem is small black window poping up in right down corner from time to time (suspecting cache over Wifi I set the cache to 20%, but with no luck).

closed 2015-03-17 by fxxer