RasPlex/RasPlex #361: Channels not working on RPi2

tudro, 2015-03-16:
Hello,

I have recently bought a RPi2 and started using rasplex. Everything works great, thank you very much for that :), except for the channels. I tried on the forums, but i couldn't find any fix for my problem.

The problem is that most of the channels don't work. I could get a few working, like the daily show or GT, but most of them don't (ign, funny or die, giant bomb, etc). Not even youtube works. If i use the andorid app or the web app (all in the same local network as the rasplex) all the channels work. I also bought the 2 codecs from rasplex, but still no luck.

For youtube it doesn't even get to the single clip page, where i can hit play. I can see the list of, for example, the most recent videos, but when i select one it says 'please wait' for a few moments and then nothing happens.

For the rest of the channels that don't work, i do get the clip page, but after i hit play, i have the 'please wait' for a few seconds then "Playback failed one or more items failed to play". 

Rasplex log: http://pastebin.com/dKmMqyXZ
Packets for a failed youtube attempt: https://drive.google.com/file/d/0B7u1EBJeM_g8R0hIMGltUVlRbFU/view?usp=sharing
Packets for a successful daily show attempt: https://drive.google.com/file/d/0B7u1EBJeM_g8LWZTU2FRUFBub3M/view?usp=sharing 
Serial: 000000002ab03491
Rpi version 1gb
SD: sony sdhc class 10
wifi
rasplex v0.5.1
Rasplex install: I copied the last version in windows on an sd card then installed it using the raspberry.
plex version 0.9.11.16
the plex server is remote
i have a good quality 5v/2a power supply and only the wifi on the usb
cpu around 3%, mem around 50%

Thank you

NedtheNerd, 2015-03-17:
This is most likely a hardware or network issue, suggest that you reflash your SD card using GetRasplex and check your local network settings.

I have installed all of the Channels you mention and all are working fine. I have '720' checked for 'Online video quality' in Preferences -> Network -> Quality.

closed 2015-03-17 by NedtheNerd

tudro, 2015-03-17:
Hello,

Thank you for the answer. I don't think it is a network or hardware issue, because a few of them work. Also, the android phone is in the same network and all the channels work.

I have tried all the settings for 'Online video quality'. As i said, for youtube it doesn't even get to the 'single clip' page.

I will try to reflash the sd card.

Thank you

tudro, 2015-03-17:
Hello,

I have reflashed the card and i still have the problem. I guess there is a problem with this version of rasplex, but hopefully the new version that is coming soon from what i understand will fix this.

Thank you

NedtheNerd, 2015-03-17:
This is not a problem with Rasplex, all the Channels you mention are working fine for me.

tudro, 2015-03-17:
Then i don't know what is wrong with them. I would understand if none of them worked, but some do, so it's not the network, nor the hardware. Strange.

NedtheNerd, 2015-03-17:
How did you install the Channels, I usually use Plex web for installing channels, then when I return to Rasplex they are fully operational?

tudro, 2015-03-17:
This is how i installed them, too.

tudro, 2015-03-19:
I have managed to get it working by placing the RPi in the same network as the PMS. Still no luck with remote PMS. Maybe there is a timeout on plex or rasplex, maybe something to do with the routers. Hard to say especially when some of them work.

I will continue to investigate and try to make it work remotely. I'll post again if i find anything.

Thank you