RasPlex/RasPlex #362: Problem with Hyperion instalation

TechaY, 2015-03-17:
Hello, 

I just buy the new RPI 2 and try to install hyperion like i did before (https://github.com/tvdzwan/hyperion/wiki/Installation-on-RPi-with-OpenELEC)

You can see here my problem :
RasPlex:~ # curl -L --output install_hyperion.sh --get https://raw.githubusercon
tent.com/tvdzwan/hyperion/master/bin/install_hyperion.sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (60) server certificate verification failed. CAfile: /etc/pki/tls/cacert.pem CRLfile: none
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

Any idea ? 

Thank's

dalehamel, 2015-03-17:
Hyperion will be supported natively in the next release.

TechaY, 2015-03-19:
Ok cool ! I will wait

Thank's

closed 2015-03-19 by NedtheNerd