RasPlex/RasPlex #364: Video with High Bitrate (RPI 2)

TechaY, 2015-03-19:
Hello,

I've got some media with high bitrate. There is still a filter in the last version (RPI2) and plex has to transcode it. I would like to know if the RPI 2 is powerfull enought now to playback this kind of media : 

Thank's 

Média

    Video Resolution 1080p
    Durée 1:01:37
    Bitrate 16326 kbps
    Width 1920
    Height 1080
    Aspect Ratio 1.78
    Container MKV
    Video Frame Rate 24p

Vidéo

    Codec H264
    Bitrate 14396 kbps
    Langue English
    Bit Depth 10
    CABAC 1
    Chroma Subsampling 4:2:0
    Color Space yuv
    Durée 1:01:37
    Frame Rate 23.976 fps

BradMeier, 2015-03-19:
The Raspberry Pi (1&2) can't natively decode videos using 10bit depth, this is a limitation of the hardware, not software.

closed 2015-03-19 by BradMeier

TechaY, 2015-03-19:
Ok ! I was hoping Raspberry Pi 2 could decode it... Thanks for the information.