RasPlex/RasPlex #365: Random restarts

topitox, 2015-03-31:
My Raspberry Pi 2 keeps restarting at random times (especially at nights when I'm not using it) without any reasons. The reference crash id is 41411.

mattstutz, 2015-04-16:
I have the same issue. The reference crash id is 59024.
I'm using Pi2 512, rasplex version 0.5.1, 32 GB SDHC class 10, WIFI connection.
Crash happens only if rasplex is not being used for some time (15 minutes, half an hour).
Using different installation on different SD Card does not solve the problem

g1t-dlanor, 2015-04-16:
The most frequent reason for RPi2 instability is an insufficient current capacity of the power supply.
Since the RPi2 has a quad-core CPU it needs considerably more current than the RPi1 models.
Please check that your PSU for the RPi2 is rated to deliver 2 Amperes of current.
Another frequent reason for instability is SD card corruption, which can occur as a result of other crashing and/or power termination without proper software shutdown.

mattstutz, 2015-04-17:
Unfortunately changing the power supply (from tv usb port to external USB power supply) did not solve the problem. Changed also the wifi adapter (different model) but also here no change, except that crash id is now 59885.
If i'm using plex, I have almost no problem, but if I change the hdmi input to different source, after 15-30' rasplex is crashing (and tv is changing input source due to reboot of rasplex).

Here the link to the plexhometheater.old.log file: http://pastebin.com/mF3RfuRt.
Any other ideas?

topitox, 2015-04-17:
@g1t-dlanor I changed from my old Rpi adapter to a 2A adapter and now I get less frequent crashes. @mattstutz how many ampers is your USB power supply?

mattstutz, 2015-04-17:
It's a 2.1A version. Also the UBS port on my TV had 2A.

g1t-dlanor, 2015-04-17:
@mattstutz:
I'm not sure how your TV works, but the majority of TV sets disable power to their USB ports when the TV set is switched off to standby mode. That will then crash any RPi powered from those ports, with high risk of SD card corruption. That risk may be lowered by having EDID protocol shut down the RPi as TV is switched to standby, but judging from your original post that's not what you've tried to do.
Still, with the stand-alone 2.1A PSU there should be no problem. Such a PSU should be able to power an RPi2 reliably. But if you haven't reflashed since you used the TV USB ports to power the RPi, you may have residual corruption from those crashes.

mattstutz, 2015-04-18:
@g1t-dlanor:

Thanks for your quick reply.

 

I flashed now another sc card and switched the RPi to permanent external PSU. Will monitor if it works now.

 

Disadvantage ist hat the RPi is not running all the time. With my original set up (having the RPi powered by the TV, special 2A USB port), the RPi had been started and stopped with/by the TV.

My TV is a Samsung 7k series. Don’t no anything about EDID protocoll, but what I observe, the TV is not just disabling power to their USB ports when switched off. My RPi seems to receive a shutdown command first before the TV is finally switched off.

mattstutz, 2015-04-19:
Unfortunatly still the same, no improvement

Zunee, 2015-05-07:
I have the same issue, which appeared to start after the update to 0.6.0.  This happened on two different pi's a pi2 and a pi 1b.   Both cases I was on a show episode and hit back and poof it rebooted.  Both have adequate power supplies and have never done this before.  

Crash Id: 72624
Serial #: 000000001b8ae91f
Pi: Pi2
SD: Class 10 (Kingston)
Network: Ethernet
Version: Latest release 0.6.0 pre-release

NedtheNerd, 2015-05-20:
Stability has been improved in next release.

closed 2015-05-20 by NedtheNerd

Zunee, 2015-05-20:
Awesome!