RasPlex/RasPlex #369: Home PIN entry ignores arrow keys

chrissearle, 2015-04-20:
Testing 0.5.1 on an RPI2.

Normal arrow/enter/esc navigation works fine either with a normal USB keyboard or via flirc.tv and an apple remote.

But - if I change to a home user that has a PIN then the arrow keys stop working (again - for both normal keyboard and flirc). To successfully enter a PIN I have to have a keyboard with numbers - so I can't enter it with the apple remote.

chrissearle, 2015-04-20:
Same issue if I connect with the iOS app as a remote. It can control the app fine - but not move around the PIN entry dialog.

NedtheNerd, 2015-05-21:
This is not specifically a Rasplex issue, seems to be a problem with Plex HT, search the forums for 'pin entry remote'.  If it is ever fixed in Plex HT then it will be incorporated into Rasplex.

Closing the issue.

closed 2015-05-21 by NedtheNerd