RasPlex/RasPlex #370: Only the first keyboard works

biltongza, 2015-04-25:
I just bought a wireless keyboard/mouse combo, I was quite disappointed when I plugged in the receiver and it wasn't working. I figured the drivers for it weren't included in 0.5.1, so I got excited when I saw 0.6.0 was released hoping the drivers might be included. 

Once I flashed 0.6.0 I used a standard wired keyboard to set it up, then once everything was settled I plugged in my receiver again, and once again the keyboard wasn't working. I SSH'd in and found that the Pi was actually picking it up as a mouse and keyboard, but rasplex still wasn't registering any keypresses. 

I noticed that the input number was incrementing every time I plugged in the receiver and thought that might have something to do with it, so I rebooted my pi with only the wireless keyboard receiver plugged in and it is now working fine. Any other keyboard I plug in now doesn't work, even if I unplug the receiver, until the system reboots.

dalehamel, 2015-04-25:
Try with both keyboards plugged in before booting.

Hotplug (adding a USB device while the system is already running) does not
always work.

NedtheNerd, 2015-04-25:
The majority of USB devices have to be connected before powering on a Pi, same issue on Raspbian with a number of devices, not a Rasplex issue, closing.

closed 2015-04-25 by NedtheNerd