RasPlex/RasPlex #371: 0.6 no sound in playback

fuzzy76, 2015-04-26:
I have sound in the menu, but none during any playback. Upgraded from 0.5.0. Tried flipping ffmpeg avio, dts and ac3 settings. Neither helped.

dalehamel, 2015-04-26:
What are you using for audio output

fuzzy76, 2015-04-27:
HDMI. It works fine in the menu, so I doubt it's a driver issue.

BradMeier, 2015-04-27:
Please ssh in to your Pi and report the output of: tvservice -a

fuzzy76, 2015-04-28:
```PCM supported: Max channels: 2, Max samplerate:  48kHz, Max samplesize 16 bits.```

That doesn't seem right... The HDMI runs through the TV which in turn is connected to my surround receiver with TOSLINK cable. But I got that output while the TV is off (I'm not home at the moment), if that makes any difference. I will make some more tests when I get home.

g1t-dlanor, 2015-04-28:
Commands related to available HDMI services may depend on many different things, all of which may be needed to obtain the correct result:

1: RasPlex should be connected to an HDMI input of the TV or receiver (obviously).
2: The TV or receiver should be active and should have the RasPlex input selected.
3: For the case of connection of RasPlex to a receiver, that receiver should be connected to the HDMI input of a TV set, and that TV set should also be active and should have the input from the receiver selected.

If any of the above conditions are not true, then the results might not relate to your normal RasPlex playback situation.

fuzzy76, 2015-04-29:
Ok, I did some more investigating. It seemed that the videos I tried all had 5.1 tracks of some sort. I tried with a video with a regular stereo audio track, and that *did* work fine.

After reading some comments in the forum I tried forcing output to "2.0" instead of "5.1", and that gave me sound on all videos. Just not with surround. :-/

tvservice -a still gives:
```PCM supported: Max channels: 2, Max samplerate:  48kHz, Max samplesize 16 bits.```

Could it be that my TV identifies wrongly? But that doesn't really explain why it used to work before the upgrade...

dalehamel, 2015-04-29:
We did make an update to the raspberry pi audio, it's possible there is a
legitimate bug.

fuzzy76, 2015-05-03:
FWIW, most other reports seem to say "center channel is missing", but since the fix is the same, I'm guessing it's a different facet of the same underlying bug.

NedtheNerd, 2015-05-20:
Should be fixed in next release.

pengc99, 2015-06-07:
Just wanted to add that I'm seeing the same problem. Worked in 0.5.1, stopped working in 0.6.1, same symptoms as OP.

Reilos, 2015-06-12:
I sometimes have no sound after i paused some video for a while, or when starting the next title. Video is playing fine. This problem goes away if i switch the TV to a TV channel and back to CEC.

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

Reilos, 2015-07-19:
I'll keep an eye out for it. However, for me it seems to be very random. I'll report back if i still encounter the issue again in 0.7.0

checkso, 2015-07-20:
I still have the same issue, even with 0.7.0 :(

tvservice -a 
PCM supported: Max channels: 2, Max samplerate:  48kHz, Max samplesize 24 bits.
AC3 supported: Max channels: 6, Max samplerate:  48kHz, Max rate  640 kb/s.

With DTS still no 5.1 sound

bramvandartel, 2015-07-20:
Nice to see 0.7.0 has been released, good job! 

The sound issue, like checkso noticed, still appears to me as well. Except at this moment I don't even have any sound selecting 2.0. I just reverted back to 0.5.1 as that appears to be the most stable release for me. 0.6.0 crashed a lot.

NedtheNerd, 2015-09-05:
Fixed in next release.

Reilos, 2015-09-07:
Nice. I forgot to report back, but i also still encounter the issue. Good to know it's going to be solved.

NedtheNerd, 2015-09-08:
Can you please check in latest version 0.7.1 and advise.

Xirixiz, 2015-09-08:
Hi NedtheNerd, unfortunately the issue persists. Tested with two Raspberry devices and on Panasonic and Philips television. 

The following will provide audio output:
- 2.0 and all other settings off in the audio section.

Tried:
- Other HDMI cable(s)
- Non ARC HDMI port

Next steps for me to try:
- Optical out via HT
- Verify TV settings
- Try at a friends Sony TV/HT

All seems to be ok since was working in releases prior to version 0.6.0.

NedtheNerd, 2015-09-08:
@Xirixiz 

Please see this topic https://forums.plex.tv/discussion/181255/rasplex-guide-to-audio-settings#latest

NedtheNerd, 2015-09-11:
Issue fixed in 0.7.1, closing issue.

closed 2015-09-11 by NedtheNerd

Reilos, 2015-09-12:
It's not fixed. I'm experiencing the same issue in 0.7.1

NedtheNerd, 2015-09-12:
Read the Rasplex - Audio Settings guide on the Rasplex Forum.

Reilos, 2015-09-12:
I dit, it was in your earlier reply. Settings are correct. Still same issue

NedtheNerd, 2015-09-14:
Can you post back result of entering tvservice -a via SSH, thanks.

Reilos, 2015-09-14:
I'm 99% sure it was equal to the the first line of that manual. I have set the settings accordingly. My TV is a UE48HU7500. I'll check again later today when i'm back home

Xirixiz, 2015-09-14:
RasPlex:~ # tvservice -a
     PCM supported: Max channels: 8, Max samplerate: 192kHz, Max samplesize 24 bits.
     AC3 supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
   MPEG1 supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     MP3 supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
   MPEG2 supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     AAC supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     DTS supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
   ATRAC supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     DSD supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
    EAC3 supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
  DTS_HD supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     MLP supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
     DST supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
  WMAPRO supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
Extended supported: Max channels: 8, Max samplerate: 192kHz, Max rate 2040 kb/s.
RasPlex:~ #

Reilos, 2015-09-14:
Just checked, it's indeed the first one from the manual. PCM supported: Max channels: 2, Max samplerate: 48kHz, Max samplesize 24 bits. Settings are also as stated in the manual.