RasPlex/RasPlex #372: PCM audio not passed through

Pipshrey, 2015-05-06:
Sample one is still uploading to dropbox at the time of this post (give it 20 mins or so)

Receiver is capable:
PCM supported: Max channels: 8, Max samplerate: 192kHz, Max samplesize 24 bits. 
AC3 supported: Max channels: 6, Max samplerate:  48kHz, Max rate  640 kb/s.
DTS supported: Max channels: 7, Max samplerate:  96kHz, Max rate 1536 kb/s.

Rasplex running on pi2 with 0.6.0 reports: 'unknown codec: PCM' and subsequently audio and video are transcoded. 

Here are 20 sec samples – one is 48.9 MB and two is 14.8 MB – of two mkv files that show the issue, allowing you to reproduce the error:

https://dl.dropboxusercontent.com/u/13997835/winter/one.mkv
https://dl.dropboxusercontent.com/u/13997835/winter/two.mkv

One is an H264 (I have the license for this and it is initialised and working) and two is mpeg2. If you need anything else, tell me.

Pipshrey, 2015-05-13:
Further to the above, and I hope it will help. The following file can direct play and pass PCM audio. So, what is different about this file and those above that do not Direct Play? Ned has been helpful (https://forums.plex.tv/index.php/topic/160106-does-rasplex-support-pcmlpcm-passthrough/ ) but between us we still have not go to the bottom of matters.

https://dl.dropboxusercontent.com/u/13997835/winter/three.mkv

Pipshrey, 2015-05-13:
I can report something knew: I can force Rasplex to passthrough and direct play a file with PCM audio; in fact I can make two.mkv (https://dl.dropboxusercontent.com/u/13997835/winter/two.mkv) pass the PCM and Direct play by doing the following:

1. ACC is the default track (as i added this to stop the transcoding) 
2. Select the PCM track manually in rasplex
3. The file now Direct Plays and the PCM audio is passed through to the receiver. 

Does this not indicate some kind of bug? It seems that if PCM audio (in some cases) is the default track it will transcode.

Kwiboo, 2015-05-13:
Should be fixed in next RasPlex release, PCM has been added to the known audio codecs list.

Pipshrey, 2015-05-13:
A lot of concerts i own have pcm audio only, so this is excellent news. Thank you.

NedtheNerd, 2015-05-20:
Fixed in next release.

closed 2015-05-20 by NedtheNerd