RasPlex/RasPlex #373: openvpn and vpn options missing

unixjeff, 2015-05-07:
openvpn and the vpn options in networking seem to be missing in 0.6.0

StuartRob, 2015-05-17:
I have the same problem, both with an update of 5.0.1 to 6.0 and with a fresh 6.0.

NedtheNerd, 2015-05-20:
OpenVPN has been removed from OpenELEC, if it ever comes back then it will be incorporated into Rasplex.  Sorry.

closed 2015-05-20 by NedtheNerd

jack0w, 2015-05-29:
Is it possible for the openvpn binary to be included and installed in future releases? At least it would be simple for a script to be run to connect, if someone so desired - no GUI would be required.

I see for openelec they've got openvpn available through a repo.

Alternatively, is there a way I can do this myself with 6.0?

dalehamel, 2015-05-29:
Curious what the use case is for running a vpn client on rasplex

jack0w, 2015-05-29:
At home I have a port opened for my own openvpn server. This is how I securely connect to my Plex server when away from home. Works great on Windows/Linux/Android/iOS and up until now, Rasplex too!

dalehamel, 2015-05-29:
If there was support for it previously then perhaps we could salvage that
work at a minimal development cost.

I do like the idea of using rasplex on the go like this

jack0w, 2015-05-29:
It would be great to see that support return! Thanks for considering this.

Here is a link to how myself and others set it up on the Plex forums - https://forums.plex.tv/index.php?/topic/109882-OpenVPN-client-access%2C-possible%3F

StuartRob, 2015-06-03:
I had openvpn running great with 0.5.1. I used it for BBC iplayer. I has a look at adding the binary to 0.6.0 but it was beyond me. I've downgraded from 0.6 to .0.5.1 because I want to use a VPN.

jack0w, 2015-06-03:
Same with me for the time being. Hopefully dalehamel will be able to restore OpenVPN in a future version

NedtheNerd, 2015-06-03:
OpenVPN is back in next release, version 2.3.5

jack0w, 2015-06-03:
Awesome! Thanks for sharing that Ned, look forward to its release :)

iAutomate, 2015-06-03:
Excellen!

StuartRob, 2015-06-04:
Great! Look forward to upgrading, I liked 0.6.1 a lot (now if only the BBC iplayer channel would work for both live and on-demand…)

NedtheNerd, 2015-06-04:
Assume that you mean 0.6.0.  The iPlayer channel issue is not just restricted to Rasplex, same with YouTube.

chownsauce, 2015-06-16:
Anyone know if the pptp binary still in 0.6.0 or is that removed as well?