RasPlex/RasPlex #374: "Watch Later" Flinging from iOS/Android to Rasplex does not work

tomkat83, 2015-05-07:
Hi guys

First things first: Awesome effort from your side! Let me know if I can help you further. 

Thanks so much, Tom

**Serial number of your pi**: 0000000014eb3164
**Crash ID, if known**: no Crash
**Size of Raspberry Pi (256, 512)**: Raspberry 2, so 1024MB
**Type of SD card (SDHC class 10?)**: SDHC class 10
**Internet connection type (Ethernet or Wifi)**: Ethernet, otherwise absolutely no issues at all streaming 1080p content. 
**Version of RasPlex**: 0.6.0
**Log files**: See http://pastebin.com/jpRVkF6Z for plexhometheater.log, I fling around 17:12
**How did you install rasplex**: gui installer, fresh (no upgrade) using Windows 7
**Best description you can give of the problem**: Flinging Watch-Later videos from either iOS or Android does not work at all. The rasplex "spinning wheel" appears, but then nothing happens. 

**Steps to reproduce (what were you doing when it happened? Can you make it happen again?)**:
1.) Start RasPlex
2.) Connect remote iOS or Android device to play remotely using RasPlex
3.) Fling any video from the queue
4.) nothing happens

**Version of Plex Server**: 0.9.12.1
**Type of plex server (NaS? Remote? Local?)**: powerfull Windows7 Server
**Power supply type and power rating ( amperage?)**: state-of-the-art 2.1A
**Comment on system load, cpu usage, and free memory (ssh in and run 'top' and 'free -m')**: Nothing suspicious

NedtheNerd, 2015-05-07:
Known issue in Plex HT, has been fixed in Plex HT1.4 so should be fixed in next Rasplex release.

tomkat83, 2015-05-07:
Awesome =). Thanks Ned

closed 2015-05-07 by tomkat83

NedtheNerd, 2015-05-07:
Just tested on next Rasplex release (based on Plex HT1.4), works fine from iOS app.