RasPlex/RasPlex #375: Video playback stops after freeze

thomasbriet, 2015-05-14:
I have, i think, a new bug in Rasplex.

Serial number of your pi: 00000000059b89d6
Crash ID - none available
Size of Raspberry Pi: 1024
Type of SD card: SDHC class 10
Internet connection type: Ethernet
Version of RasPlex 0.6.0

How did you install rasplex? Via the installer on MacOSX 10.10.3
Best description you can give of the problem:

After a few minutes of normal playback. The screen freezes, the subtitles continue and after a few seconds Rasplex jumps back to the movie menu. It remembers the play location.

Steps to reproduce

1. Watch movie or serie
2. Wait a few minutes
3. Screen freezes

Here is my Log.
http://pastebin.com/fFuTVGu4

I'm not sure what to look for, but it happens on all movies or series i'm currently watching.

SkebbZ, 2015-05-18:
I have the same issue after updating to new server software, running on Windows. Worked fine on Android clients though. Using Rpi2,rasplex  0.6

NedtheNerd, 2015-05-18:
This is not a bug, as we would have had thousands of issues filed or posts on the forum.  

This could be:

a. Possible SD card corruption.
b. Power supply issue, must be at least 2A or better.
c. USB cable of inferior quality.

Suggest checking power supply, USB cable, reflashing card, in that order.

Regards

thomasbriet, 2015-05-18:
I think i "fixed" it by setting "Use LCD/FVD" unchecked in advanced settings.
No more freezing and returning to summary screen so far. (10 hours of watching).

NedtheNerd, 2015-05-18:
This is unchecked by default, you obviously checked it and gave yourself the problem.

Glad it's 'fixed'.

Closing issue.

closed 2015-05-18 by NedtheNerd