RasPlex/RasPlex #376: No playback

Kroganil, 2015-05-17:
Since 0.6.0 upgrade i can't playback (full screen of artifacts) some mkv, but 0.5.1 was ok with those files. The playback looks like when you try to start old NES cartrige but it dosent work.

Mediainfo:

General
Unique ID                   : xxx
Complete name               : Z:\xxx\xxx\xxx - SxxExx - xxx.mkv
Format                      : Matroska
Format version              : Version 4 / Version 2
File size                   : 1.29 GiB
Duration                    : 41mn 13s
Overall bit rate            : 4 471 Kbps
Movie name                  : xxx
Encoded date                : UTC 2015-05-10 21:53:17
Writing application         : mkvmerge v6.7.0 ('Back to the Ground') 64bit built on Jan  8 2014 14:31:51
Writing library             : libebml v1.3.0 + libmatroska v1.4.1

Video
ID                          : 1
Format                      : AVC
Format/Info                 : Advanced Video Codec
Format profile              : High@L3.1
Format settings, CABAC      : No
Format settings, ReFrames   : 2 frames
Codec ID                    : V_MPEG4/ISO/AVC
Duration                    : 41mn 13s
Bit rate                    : 4 190 Kbps
Width                       : 1 280 pixels
Height                      : 718 pixels
Display aspect ratio        : 16:9
Frame rate mode             : Constant
Frame rate                  : 25.000 fps
Color space                 : YUV
Chroma subsampling          : 4:2:0
Bit depth                   : 8 bits
Scan type                   : Progressive
Bits/(Pixel*Frame)          : 0.182
Stream size                 : 1.21 GiB (94%)
Title                       : xxx
Language                    : French
Default                     : Yes
Forced                      : No
Color primaries             : BT.709
Transfer characteristics    : BT.709
Matrix coefficients         : BT.709
Color range                 : Limited

Audio
ID                          : 2
Format                      : AC-3
Format/Info                 : Audio Coding 3
Mode extension              : CM (complete main)
Format settings, Endianness : Big
Codec ID                    : A_AC3
Duration                    : 41mn 13s
Bit rate mode               : Constant
Bit rate                    : 192 Kbps
Channel(s)                  : 2 channels
Channel positions           : Front: L R
Sampling rate               : 48.0 KHz
Bit depth                   : 16 bits
Compression mode            : Lossy
Stream size                 : 56.6 MiB (4%)
Title                       : xxx
Language                    : French
Default                     : Yes
Forced                      : No

Crash Id: n/a
Serial #: ?
Pi: Pi2
SD: Class 10 (Sandisk extreme)
Network: Gb Ethernet
Version: 0.6.0 pre-release
P.M.S.:FreeNAS-9.3-STABLE-201504152200, pms: 2.4.4

NedtheNerd, 2015-05-21:
Not sure why this is happening, the file should play OK, have you tried reflashing the SD card?

Kroganil, 2015-05-24:
Yep, and it doesn't work.

g1t-dlanor, 2015-05-24:
@Kroganil: When you attempt playback and get this 'garbled screen', does everything else work normally ? By this I mean whether or not the program still responds to keyboard commands.

If it does still respond, please try using the command key "i" from a USB keyboard (or corresponding remote key if you have such). This should display various information about the media on top of the rendered video, and hopefully that will work even when the video playback is garbled. One of the things this should reveal is whether or not the stream is being sent to the RPi as 'Direct Play' or by transcoding, the latter being something I doubt whether your FreeNAS server can handle. This and other information displayed could give us a clue to understanding the problem.

Btw: There's nothing wrong with the info you already sent above, but the method I outlined will give us the information as seen by the PMS server and the PHT client in RasPlex, so we can compare this to the MediaInfo data you already sent. Any significant difference between them would indicate an issue to focus our attention on.

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

NedtheNerd, 2015-09-08:
No feedback from OP, closing issue.

closed 2015-09-08 by NedtheNerd