RasPlex/RasPlex #377: crash report loop

xortim, 2015-05-19:
My TV turned off and the CEC Device name was "Recorder." After rebooting I received what appeared to be an infinite loop of "Crash Report" prompts referencing incrementing bug report ids (78060+).

Is there anyway to disable the crash report system?

I'm in a bit of a hurry at the moment I'll reply with some more useful details later.

NedtheNerd, 2015-05-20:
This could be a SD card corruption, suggest reflashing.

xortim, 2015-05-21:
I'll try reflashing tonight. Until then, there's the device details
Pi Model: B v2
Serial: 000000001a63b942
rasplex Version: 0.6.0
Type of SD card: class 10
Internet connection type: Ethernet 
Installed via dd of img file from Linux

xortim, 2015-06-03:
Reflashing the SD card using the installer appears to have corrected this. I'll reopen another and reference this if it crops up again. Thanks!

closed 2015-06-03 by xortim

jgeusebroek, 2017-01-12:
It's a bug and NOT a currupted SD card (at least not in my case). Flashing is not necessary, all you have to do is empty the `/storage/.plexht/temp/CrashReports/*` directory. 

SSH to rasplex and perform `rm -rf /storage/.plexht/temp/CrashReports/*`