RasPlex/RasPlex #378: Rasplex not responhding to key press via CEC

norbusan, 2015-05-19:
I am having the following setup: Rasplex 0.6.0 on Raspberry II, connected via HDMI to Sony TV. Sony's BraviaLink is turned on, and sends key press from the remote to rasplex.

I can see the key press even in the log files:
`````
08:29:57 T:1866462288   DEBUG: CecLogMessage - >> Free use (E) -> Recorder 1 (1): user control pressed (44)
08:29:57 T:1866462288   DEBUG: CecLogMessage - SetCurrentButton down (2) D:0ms cur:2
08:29:57 T:1866462288   DEBUG: CecLogMessage - key pressed: down (2) current(ff) duration(0)
08:29:57 T:1866462288   DEBUG: CecLogMessage - Changed key down (2) D:0ms cur:ff
08:29:57 T:1866462288   DEBUG: CecLogMessage - key pressed: down (2, 0)
08:29:57 T:1866462288   DEBUG: PushCecKeypress - received key a7 duration 0
08:29:57 T:1866462288   DEBUG: CecLogMessage - CheckKeypressTimeout T:299.269
08:29:57 T:1866462288   DEBUG: CecLogMessage - Key down: idle (duration:0) (2) timeout:500ms (rel:500,rep:0,prs:1,rel:0)
08:29:57 T:1866462288   DEBUG: CecLogMessage - >> e1:8b:02
08:29:57 T:1866462288   DEBUG: CecLogMessage - >> Free use (E) -> Recorder 1 (1): vendor remote button up (8B)
08:29:57 T:1866462288   DEBUG: CecLogMessage - key released: down (2) D:218ms
08:29:57 T:1866462288   DEBUG: PushCecKeypress - received key a7 duration 218
08:29:58 T:1799353424   DEBUG: CecLogMessage - << Recorder 1 (1) -> TV (0): POLL
08:29:58 T:1799353424   DEBUG: CecLogMessage - << 10
08:29:58 T:1799353424   DEBUG: Previous line repeats 1 times.
08:29:58 T:1799353424   DEBUG: CecLogMessage - >> POLL not sent
`````
but there is no reaction from the PHT/Rasplex GUI at all.

I tried resetting my storage area, without change.

Thanks

Norbert

NedtheNerd, 2015-05-20:
Can I suggest that you try refreshing the SD card.

norbusan, 2015-05-21:
Tried that now, same effect. Grabed the 0.6.0 release for RP2, flashed it to my SD card, installed it into the RP2, rasplex booted, enlarged the fs, rebooted, usual setup procedure.

Then I actived debug logging and restarted rasplex, logged per ssh into the device, tail -f /storage/.plexht/temp/plexhometheater.log and I get similar events as above:
`````
15:34:26 T:1858073680   DEBUG: CecLogMessage - >> Free use (E) -> Recorder 1 (1): user control pressed (44)
15:34:26 T:1858073680   DEBUG: CecLogMessage - SetCurrentButton down (2) D:0ms cur:2
15:34:26 T:1858073680   DEBUG: CecLogMessage - key pressed: down (2) current(ff) duration(0)
15:34:26 T:1858073680   DEBUG: CecLogMessage - Changed key down (2) D:0ms cur:ff
15:34:26 T:1858073680   DEBUG: CecLogMessage - key pressed: down (2, 0)
15:34:26 T:1858073680   DEBUG: PushCecKeypress - received key a7 duration 0
15:34:26 T:1858073680   DEBUG: CecLogMessage - CheckKeypressTimeout T:161.300
15:34:26 T:1858073680   DEBUG: CecLogMessage - Key down: idle (duration:0) (2) timeout:500ms (rel:500,rep:0,prs:1,rel:0)
15:34:27 T:1858073680   DEBUG: CecLogMessage - >> e1:8b:02
15:34:27 T:1858073680   DEBUG: CecLogMessage - >> Free use (E) -> Recorder 1 (1): vendor remote button up (8B)
15:34:27 T:1858073680   DEBUG: CecLogMessage - key released: down (2) D:211ms
15:34:27 T:1858073680   DEBUG: PushCecKeypress - received key a7 duration 211
15:34:27 T:1799095376   DEBUG: CecLogMessage - expected response not received (90: report power status)
15:34:27 T:1799095376   DEBUG: CecLogMessage - << 10:8f
15:34:27 T:1923085392   DEBUG: Error: Requested setting (videoscreen.textures32) was not found.  It must be case-sensitive
`````

Anything else I can try? It looks strange that the Cec logs receiving a key press, but the rasplex doesn't react?

norbusan, 2015-05-21:
BTW, is there any chance that dropping something into `/storage/.plexht/userdata/keymaps/` might change something?

NedtheNerd, 2015-05-21:
Yes, it definitely will, any xml file in their will overwrite the default and if incorrect will stop CEC working, remove the file, reboot and see if this issue still presents itself.

norbusan, 2015-05-21:
Hi
sorry for the misunderstanding, there is *no* file there, I thought by dropping one in there it might go back to work ;-)

NedtheNerd, 2015-05-21:
I have a Sony Bravia TV and CEC works out of the box, check that CEC is enabled.

norbusan, 2015-05-22:
Hmm, it looks like enabled, otherwise the keypresses would not *arrive* in rasplex, or??? Why are they mentioned in the rasplex log file? That means the CEC is connected and enabled, as far as I understand.

NedtheNerd, 2015-06-05:
It just means that Rasplex is confirming that a CEC signal has been generated and is being sent to the TV, for some reason your TV is not either receiving the signal or is not processing it.  That fact that you have the entry in the log file proves that this is not a Rasplex issue.  Have you checked the Bravia Sync settings on your TV.  Try swapping the HDMI cable, some are not that good, especially ones that are less than HDMI 1.4 spec. If this really was a Rasplex issue the forums would be full of complaints.

Now closing issue.

closed 2015-06-05 by NedtheNerd

norbusan, 2015-06-09:
Hmmm, I don't understand your answer. Why should the *TV* react?

My interpretations is:
I press button on remote -> TV receives the signal -> TV forwards the signal via HDMI to rasplex -> Rasplex receives the signal -> Rasplex reacts (scroll up, down etc).

Now, I can confirm that the information is *received* on the rasplex side, as clearly shown on in the log files. So there is nothing more the TV is supposed to do. Rasplex needs to act on the pressed *and* received buttons.