RasPlex/RasPlex #379: RasPlex Sign In to Plex doesn't work

EPICmynamesBG, 2015-05-21:
Serial: 00000000d9b7d821
Crash ID: N/A
Size of Pi: 1GB (Raspberry Pi 2)
Memory Card: Sandisk Micro-SDHC 16GB
Network Connection: Wireless via EDIMAX USB Wi-Fi dongle
RasPlex Version: 0.6.0
Log Files: Was unable to turn on network logging. Would hit enter to check mark, but would not check mark.
Install method: Mac OS X GetRasPlex installer
Description:  So excited to install RasPlex onto  my Pi 2 today. Got everything set up, and then I couldn't sign into my Plex account meaning I couldn't access my LAN Plex Server. I am connected over Wi-Fi via an EDIMAX USB dongle if that makes a difference. Upon selecting Sign In, it would go to the PIN page and say requesting pin continuously, but never pull a pin. I tried the manual login, but after everything was entered and the login button was pressed, again, it just spun forever, never getting anywhere. The Wi-Fi dongle activity light was blinking, indicating something was going on.
Reproduction Steps: Install via OS X as RasPi2. Boot RasPi2 and go through setup, skipping network connection initially. After at PHT screen, go to system settings, enable Wireless networks and connected to home wireless. Attempt to log in to Plex account in order to get access to Plex Server, nothing.
Plex Server Version: 0.9.11.16 on Local (LAN) 64-bit Ubuntu Server

I will try to watch this issue, so if any more information may be needed, just let me know. Thanks for your help and all the hard work!

NedtheNerd, 2015-05-21:
Have you tried logging in to Plex in Preferences->System->Plex->Sign in rather than using the Sign in from the Home Screen?

NedtheNerd, 2015-05-21:
Incidentally, you don't need to sign in to Plex to view local libraries, if you are not seeing your local libraries but only Search and Watch Later then you have a network issue, try connecting to an Ethernet connection.

EPICmynamesBG, 2015-05-21:
Yes, I did try signing in through that other path, same result. And because of how I have configured my server (because most of the year I am on a large college campus), it does require a device sign in to see the server on LAN. But I will try Ethernet to see if that changes anything.

EPICmynamesBG, 2015-05-22:
I have just tested with this with an Ethernet connection, same results: it continually spins when trying to pull a PIN or when doing the manual log in.

NedtheNerd, 2015-05-22:
Cannot suggest anything else, this is not a Rasplex issue but an issue with how you have configured your network/server.  Suggest that you try raising this in the forum giving full details of how you have configured your network/server, someone may be able to offer some advice.  We have tens of thousands of Rasplex users who don't have this issue.  Sorry.

Closing this issue.

closed 2015-05-22 by NedtheNerd

tppinheiro, 2015-05-26:
I have exactly the same problem. I managed to sign in into Plex, but since at least yesterday, I am unable to do so anymore. It does not auto login and if I try any of the login methods (PIN or Manual) it just spins and it neither offers a PIN nor authenticates, respectively.

Raspberry Pi B+
Network Connection: Wired
RasPlex Version: 0.6.0
Install method: Mac OS X GetRasPlex installer

dalehamel, 2015-05-26:
Can you confirm if you are behind any sort of proxy?

Can you confirm that your timezone is set correctly, and that the clock has
the right time in rasplex?

It's possible that there's an issue negotiating the HTTPS connection if the
clock is too far out of sync.

-Dale

tppinheiro, 2015-05-26:
Hi. Thank you for your reply.
I could not troubleshoot and solve the issue, so I performed a hard reset, ran the initial wizard again and it worked. That is the reason why I deleted my comment here.
Nevertheless, once again, I would like to thank you for your help. :)

dalehamel, 2015-05-26:
No problem.

I am curious if multiple people are experiencing the issue. My instinct is
that it must be clock drift or something to do with proxying.

I have actually seen the symptoms described here before during development,
when the SSL certificates were not installed correctly. That is what makes
me think it is HTTPS related, and common issues there would be proxying or
time offset.

-Dlae

EPICmynamesBG, 2015-05-26:
I was able to resolve this issue by wiping and reinstalling RasPlex to my SD card. I'm not sure if that helps anyone though.

alias1, 2015-07-19:
Had this same issue. I cancelled out of the initial setup by accident. Sat spinning forever on the sign in screen. Manually added a NTP server, then tried the sign in again and it worked fine.

Is there a way to detect the SSL error on the sign in screen and provide better feedback to the user maybe?

snajjar, 2015-07-27:
I had the same issue here. 

The problem was that the auto-detected configuration used for the default connection on network (eth0) had no DNS defined in it. I changed the network mask to 255.255.255.0, added a DNS nameserver (8.8.8.8), and Rasplex was able to connect to plex again. Try this before reformating.

dreyl, 2015-09-20:
also had the same issue..

changed the preferences for network to manually specify server.
and the problem was solved.

abak, 2016-01-20:
Same here with version devel-20151128150225-gdbb11b7

It apperas that my dhcp is proving ntp servers (from pool.ntp.org), Rasplex received those, but they were not used unless I manually specify them in the network settings.

NedtheNerd, 2016-01-20:
You shouldn't be using Dev images, load up latest release 1.0.1.

abak, 2016-01-20:
I got the image through [that binary](https://github.com/RasPlex/rasplex-installer/releases/download/1.0.1/GetRasplex-1.0.1.dmg), that I found on [that page](http://www.rasplex.com/get-started/rasplex-installers.html). It was the image that got downloaded when i asked for a Raspberry Pi 1/v1.0.1 image.

Do you want me to reproduce/create another ticket? If you point me to the source for the downloader I can also look into it.

NedtheNerd, 2016-01-20:
Have just used GetRasplex to download an image for Raspberry Pi/1.0.1 and the file downloaded is the correct one - RasPlex-RPi.RP-1.0.1.img.gz not the file you quoted (devel-20151128150225-gdbb11b7).

abak, 2016-01-21:
This is indeed the name of the file that the installer downloaded on my machine (md5 ba4ba35a4cc1ddfe72ebb52f3ab9d363)
However, once flashed and booted, Rasplex reports the version number quoted above under System Settings or something.