RasPlex/RasPlex #380: Occasional Black 'flicker' whilst streaming

dankellys, 2015-05-23:
Serial number: Rasp Pi 2 Model B 1GB (Serial unknown)

Rasplex version: Latest from http://www.rasplex.com/get-started/rasplex-installers.html using Windows Gui installer

Plex media server version: 0.9.12.1.1079-b655370

Description of the bug: During playback, occasionally, the screen goes completely black, audio and video stops for around 1 second, then playback resumes. Once playback resumes, my TV indicates I am using HDMI3 (which is the connection used by the Pi).  However, it normally only displays this if you connect to a new source, or a device connected to that source boots up.  So it suggests the Pi is not sending signal via HDMI for that split second, then when it starts again, the TV recognises a device.

NedtheNerd, 2015-06-08:
Have you tried changing the HDMI cable?  Cheap or old cables can cause this issue.

NedtheNerd, 2015-07-19:
Can you please check this issue in the latest release 0.7.0 (released today) and advise.

NedtheNerd, 2015-07-28:
No further response from OP, closing issue.

closed 2015-07-28 by NedtheNerd