RasPlex/RasPlex #382: Enhancement: Enable SSL support from latest plex home theater release

Fmstrat, 2015-06-05:
Plex has finally implemented SSL into their underlying structure including PHT: https://blog.plex.tv/2015/06/04/its-not-easy-being-green-secure-communication-arrives/

This ticket is to request an update to RasPlex to include the latest PHT to fix the security issues reported here: https://forums.plex.tv/index.php/topic/101886-proof-of-concept-token-exploit-please-fix-this-massive-security-hole/

Apologies for the lack of guideline information, but I don't think the requested information was relevant for this request. However, I wanted to create this ticket since I know many of the people posting in the POC exploit thread on plex.tv are RasPlex users and they may wish to track progress.

Thank you.

NedtheNerd, 2015-06-05:
Already done in next release.

closed 2015-06-05 by NedtheNerd

Fmstrat, 2015-06-05:
Annnnd perfect. Thanks!