RasPlex/RasPlex #383: Running A Build

iAutomate, 2015-06-05:
I am trying to create a build using the following instructions provided in the Wiki (https://github.com/RasPlex/RasPlex/wiki/Native-development-environment-setup ) and not having much luck.  I resolved many errors but thinking that maybe the instructions are outdated.  One thing comes to mind is how to build for RPi vs RPi 2.

I'd like to be able to build the latest rasplex-dev-0.6.X branch.

Using the instructions and with my reference fixes to sources. I am now getting the following error:

find: `build/lib/ffmpeg/ffmpeg/lib/': No such file or directory
/home/alain/Dev/RasPlex/RasPlex/OpenELEC.tv
cp: cannot stat '/home/alain/Dev/RasPlex/RasPlex/OpenELEC.tv/build.rasplex-RPi.arm-devel-release/plexht-wip/plex/Resources/system/*': No such file or directory
make: *** [release] Error 1

Any insight would be appreciated.

Thanks, AlAiN

NedtheNerd, 2015-06-05:
The source only goes on GitHub when it becomes a stable release, currently that is for 0.4.1 which is the current stable release and that is only for the RPi1. This has been mentioned in the forums.

There are development sources that are only available to the team.

Following the wiki you should be able to build that release, once successful you can then start changing code, pull in different public sources and rebuild.  A word of warning the initial build can take up to 18 hours on a moderately powerful PC/Mac, don't even think about using a Pi to build, too slow.

Any specific reason you want to undertake a build?

iAutomate, 2015-06-05:
Thanks so much for your response.  Sorry that I could not find definitive answer on the forums.  

I was hoping to create the latest branch so that I could get VPN back.  This might not be an issue with HTTPS being released but then how long would I have to wait for a build that includes that for RasPlex.  If I can build my own dev builds I wouldn't have to wait.

I followed the instruction on the wiki and it does not work.  If you look at the issue I closed earlier (#381)  there are many missing references and access to  http://sources.rasplex.com/ is not available.  The error above is after fixing some issues and using the wiki instructions.

Thanks again.

NedtheNerd, 2015-06-05:
VPN is back, with restored system settings, in the next release we're hoping to get out 'in short order', no messing about with command line, just tested it it works like it used to.

I'll close this issue for now.

closed 2015-06-05 by NedtheNerd