RasPlex/RasPlex #385: HDMI only works in Safe Mode

adinunzio10, 2015-06-13:
Serial number of your pi:

00000000a62af8d7
Crash ID, if known - you can get this from the debug log if a crash was sent:

N/A
Size of Raspberry Pi (256, 512): 

Raspberry Pi 2 (1GB RAM)

Type of SD card (SDHC class 10?):

SanDisk Ultra 64GB MicroSDXC Class 10
Internet connection type (Ethernet or Wifi):

Ethernet
Version of RasPlex (if not the latest, please see if the problem exists on latest first): 

0.6.0

Log files (ensure that you enable "debug logging in Preferences -> System -> advanced" and try to reproduce the issue):

How did you install rasplex? (gui installer, win32 disk imager, what platform OSX/Win/Linux?): 

win32 disk imager, windows 8.1

Best description you can give of the problem:

i see splash screen ,but after that black screen, audio works. I tried enabling HDMI safe mode and that fixes my issue, but was told to report it anyway, and it doesnt look as good. My very first time setting up RasPlex I had no problems and it looked awesome, but after reboot it only works if i go into hdmi safe mode. 

Steps to reproduce (what were you doing when it happened? Can you make it happen again?):

yes and see above,  I have also reformatted multiple times. 

Version of Plex Server:

0.9.12.3.1773-937aac3

Type of plex server (NaS? Remote? Local?): 

Local off my desktop on same connection as the pi

Power supply type and power rating ( amperage?): 

using my note 4 charger, which is 5v 2a

Comment on system load, cpu usage, and free memory (ssh in and run 'top' and 'free -m'):

Mem: 269032K used, 487008K free, 0K shrd, 10301680K buff, 10301728K cached
CPU:  2.5% usr  2.5% sys  0.3% nic 86.7% idle  7.7% io  0.0% irq  0.0% sirq
Load average: 0.85 0.28 0.10 3/111 399

RasPlex:~ # free -m
             total         used         free       shared      buffers
Mem:           738          262          475            4           29
-/+ buffers:                232          505
Swap:            0            0            0

g1t-dlanor, 2015-06-13:
If I were you I would try some other SD card. Not that I consider this a prime suspect for the errors, though it is possible (SDXC is still a fairly new standard, and different manufacturers might not implement it identically). But the main reason is that you are wasting both money and storage space by using a 64GB SDXC card for RasPlex, which only uses less than 4GB. And I don't think you gain any speed either, compared to good Class10 UHS-1 SDHC cards. And such cards from multiple manufacturers are well tested since a long time, thus eliminating any suspicions of card incompatibility. (I realize that this probably won't solve your problem, but it's something worth trying anyway.)

NedtheNerd, 2015-06-13:
Is the Pi connected to a TV or monitor, if monitor is it DVI or HDMI input, if TV have you tried changing the HDMI cable (some cheap or less than HDMI 1.4 certified cables can cause this issue).

adinunzio10, 2015-06-13:
Tv, hdmi, I'm using amazon basics cable but I will try some different
ones.. Odd that it worked fine the first time I booted up. But not since. I
ordered a second pi and a new SD card that's smaller. Going to dedicate
that to rasplex and hopefully that works.

adinunzio10, 2015-06-13:
Thank you for the advice. Just got the pi the other day and didn't order a
SD card. So don't have another to test. But I will in a few days to give it
a try.

NedtheNerd, 2015-06-16:
Have you tried taking out the hdmi_safe=1 and inserting  hdmi_ignore_edid=0xa5000080?

adinunzio10, 2015-06-17:
So I tried on my monitor instead of TV and no problems. As my second Pi is for my parents, most important is that it works on their TV, so going to try that next. 

@NedtheNerd 

I just tried that now, and still no display on my TV (my TV works fine with every other raspberry pi OS I tried)

closed 2015-06-17 by adinunzio10