RasPlex/RasPlex #386: 0.6.0 and 0.5.1 do not boot

matiasw, 2015-06-15:
Hi,
So I tried installing RasPlex today. I got the GUI Installer (`GetRasplex.bin`) and used it to download first version 0.6.0 and, when that didn't work, 0.5.1. I wrote the images to an SD card that used to work fine with Raspbian before. 
Going on advice from comments on #156, I added `noram nosplash progress` to the command line. When this had no effect, I added `debugging`. Nothing, no sign of a signal (composite video) and no device shown to be connected to my router (wired connection, also no LNK light on the Pi). I'm using the original Raspberry Pi. 
If there's anything else I can do to help debug this issue, let me know!

g1t-dlanor, 2015-06-15:
If you are trying to use the composite video signal then you must edit the "config.txt" file in order to enable composite signal output. The default directives in that file enforce HDMI output.

Depending on what you are trying to do there are also some directives you may need to add to that file, in order to choose the proper composite mode (NTSC/PAL colour modulation etc.)
But the primary fix to get any output is to remove/'comment out' the entry "hdmi_force_hotplug=1", as it is this that enforces use of HDMI, regardless of whether or not an HDMI connection is detected by the hardware at power-on.

matiasw, 2015-06-15:
Thanks. With the help of [the Wiki article](https://github.com/RasPlex/RasPlex/wiki/SD-TV-Output), I was able to get output on my screen from RasPlex 0.6.0. However, it seems to be stuck in a reboot loop. Even though I still have `nosplash progress debugging`, what I see on the screen is the splash image.
Is there a log file somewhere?

g1t-dlanor, 2015-06-15:
I would try reverting the other changes you made, keeping only the changes required to get the composite output working. Possibly one of those commandline options is interfering with normal RasPlex booting, since we never use those in our own testing.
I believe that your original problem of 'not booting' was in fact just a misdiagnosis, with RasPlex booting fine except for using the HDMI output, leaving the composite signal you needed blank.
So the other measures you took to 'make it boot' were probably not needed or appropriate.

matiasw, 2015-06-15:
Well, I would say that the device still "does not boot": the bootstrap sequence is started, alright, but does not finish successfully. And, I've been unable to spawn the debug shell with command line option `debugging`.
In any case, I reverted the changes made to `cmdline.txt` and tried to boot, but it did not help. I've not made any other changes, except expanding the storage file system.

NedtheNerd, 2015-06-15:
As it seems that all the changes you have done have not resolved the issue and in fact could have compounded you issue, and that the original issue would have been solved by adding parameters to enable composite booting I would suggest that you reflash the card, only add the parameters for composite output and try that.  Rasplex expands its file system on first boot, all of the errors from the log you have quoted are normal for first boot for a non CEC display.

matiasw, 2015-06-16:
So I started fresh with an image of 0.6.0. I only changed the composite output lines as per the Wiki. The situation remains the same: unable to boot.
http://pastebin.com/DBDhk2MY the complete plexhometheater.log, after one boot attempt
http://pastebin.com/wT4mV0mf plexhometheater.old.log
The file system did get expanded correctly.

NedtheNerd, 2015-06-16:
The fact that you have a log file and an old log file proves that Rasplex does boot and is running, but because you have no display you cannot complete the first run configuration, therefore the title of this issue is incorrect.  I would suggest that you connect your Pi to a display with an HDMI input to confirm that this is a failure of the Pi to register a display using the parameters you have used for a composite display, I an assuming that you have the correct connector the RPi1 B+ and RPi2 use different composite connectors to the original RPi, see here http://www.raspberrypi-spy.co.uk/2014/07/raspberry-pi-model-b-3-5mm-audiovideo-jack/.

Suggest also flashing a card with Raspbian and checking whether this gives you a display output.

matiasw, 2015-06-16:
Should RasPlex have SSH enabled by default? At least it should get an IP address from my router via DHCP, no? I see no such activity.
I am using the original Pi, which has a dedicated composite video out. Unfortunately, I do not own an HDMI display, but I suppose I'll look for one to test on. I don't know if this is "a failure of the Pi to register a display using the parameters you have used for a composite display", because I do get image to my TV via composite video.

NedtheNerd, 2015-06-16:
Yes, SSH is enabled via default, are you using wired or WiFi connection?

matiasw, 2015-06-16:
Wired. Even the link layer is not working, meaning the LNK LED on the Pi does not light up.

NedtheNerd, 2015-06-16:
So it's an original RPi Model B, is it a 512Mb one, Rasplex will not boot on the very first 256Mb Model B's or Model A's.

matiasw, 2015-06-16:
How do I find that out?

NedtheNerd, 2015-06-16:
Check the chip on the Pi, if it has 2G in the number, it's 256MB (i.e. 2Gbit) if it has 4G in it's number, it's 512MB (4Gbit)

Maybe something like H9TKNNN2GDMP if it's a 256 (this example is for the Hynix ram, but you get the drift).

matiasw, 2015-06-16:
Ok, so it's [http://www.samsung.com/global/business/semiconductor/product/mobile-dram/detail?productId=7611](a Samsung chip, K4P2G324ED), which is 2Gb. Does that mean my device is unsupported?

NedtheNerd, 2015-06-16:
Yes, As I said Rasplex will not run on 256mb Pi's.

I'm closing the issue.

closed 2015-06-16 by NedtheNerd

matiasw, 2015-06-16:
Alright, well thanks for your time. May all of you RasPlex contributors be well & happy.