RasPlex/RasPlex #390: Audio airplay not supported anymore

fabienheureux, 2015-07-01:
Hi everyone, 
I'm a rasplex' user since quite a long time for now, and I know that you encounter many issues about Airplay streaming (audio only) implementation.
But it was working quite well in previous build, and in .6, it doesn't work anymore. 
I get a message saying something like (I'm not in front of my rPi right now..sorry, I'll edit my message when I'll be back home): this Airplay device doesn't support streaming.

Does any of you encounter the same issue ? 
I'd be willing to help debugging this issue, so if one of rasplex developper would like to point me to the part of code involved into Airplay implementation, it could be great !

Thanks

NedtheNerd, 2015-07-02:
Airplay has never worked reliably since iOS 7, it does work from time to time - no pattern, and it only ever supported audio.

I've had it working in 0.6.0 sometimes!

Waiting for Plex Inc to fix PHT.

Closing this issue as there is nothing we can do.

closed 2015-07-02 by NedtheNerd

fabienheureux, 2015-07-02:
Thanks for your reply, I thought it could be somehow related to openelec. 
I'll think about buying a dedicated rPi in order to get an audio airplay receiver

sticilface, 2016-03-06:
same here.  crashes every time i connect.  crashed 1202.  rp2.

nathanbirrell, 2016-07-15:
Any word on this @NedtheNerd ?

falahati, 2016-09-02:
Yep, still same with iTunes on El Captain :|