RasPlex/RasPlex #393: Forums & Chat Down

jhg03a, 2015-07-13:
I was trying to learn more about rasplex, and the links on the main page to the rasplex forum and chatroom appear to be out of service.

dalehamel, 2015-07-13:
The plex forums were recently hacked and will be back soon.

The chat rooms are more or less defunct for now, working to bring them back.

NedtheNerd, 2015-07-13:
Forums back up, issue closed.

closed 2015-07-13 by NedtheNerd

Conradical314, 2015-08-21:
Minor link updates in the website relating to this issue:

The link to the forum from the RasPlex [How to Help](http://www.rasplex.com/contribute/how-to-help.html) page only takes you to the Plex Forums home page.

It uses the old link of https://forums.plex.tv/index.php/forum/126-rasplex/, but it needs to be https://forums.plex.tv/categories/rasplex to have you land in the right place.

I suggest that the chat link should be removed from the [How to Help](http://www.rasplex.com/contribute/how-to-help.html) page as well, if it is defunct. Or a note saying that the room is not operating at this time