RasPlex/RasPlex #394: x265 HEVC codec support

jhg03a, 2015-07-13:
Is there a plan to get directplay support for x265 (HEVC) codec support in rasplex for the raspberry pi 2?  I'm interested in the significant space savings it can give.

NedtheNerd, 2015-07-13:
Eventually, no time frame at the moment.

MattPark, 2015-08-28:
+1 This is apparently becoming a codec of choice

KrunchMuffin, 2015-09-20:
+10

BradMeier, 2015-09-20:
Neither the RPi1 or 2 has hardware decoding for HEVC, RPi2 may be able to scrape together 720p25 in software, but don't hold your breath, will work via transcoding in PMS though.

KrunchMuffin, 2015-09-20:
bummer

closed 2015-09-21 by jhg03a

Ericmas001, 2016-10-30:
What about pi3?

g1t-dlanor, 2016-10-31:
To my knowledge the GPU and all software for RPi3 (including firmware and kernels) is identical to that of the RPi2, except for some obvious hardware changes (such as internal WiFi). So even on the RPi3 HEVC will have to be handled by software decoding, which places a severe limit on the max bitrate it will support. That bitrate will be slightly higher, due to the higher clock rates, but HEVC in good quality  1080p and above will always require server-side transcoding. As for HEVC becoming  a 'codec of choice', this is obviously a choice made only by those who care more about video compression rates than they do about media player compatibility.