RasPlex/RasPlex #396: RasPlex Release dates

anotherbhav, 2015-07-19:
Does anybody else see all the release dates for RasPlex as April 13, 2014?

If I go here, I see every release date, including the latest 0.7.0, as 'on Apr 13, 2014':
https://github.com/RasPlex/RasPlex/tags

This is not the case for rasplex-installer, those dates look correct:
https://github.com/RasPlex/rasplex-installer/tags

Awesome job btw. RasPlex is great.

Bhav

dalehamel, 2015-07-19:
This appears to be a github bug, I noticed it quite a while ago, but it's
not really major enough for me to really care.

We have a record internally of when each release is published.

KrunchMuffin, 2015-08-21:
good to know. I thought the project was dead and was going to post that question.

closed 2015-08-21 by NedtheNerd