RasPlex/RasPlex #397: Remote control numbers not working for Plex Home User login

Reilos, 2015-07-19:
Using CEC. The numbers on the TV's remote control cannot be used anymore to type the login code for a Plex Home user. Only on the screen a numpad is shown, for which you can use the remote control arrow and OK keys to enter the code. This is of course totally undesired, as now anyone can see the code you are entering on the screen. RPI2 and a Samsung UE48HU7500.

yannlet, 2015-07-22:
Same probleme for me with a samsung PS63C7700 and RPi2.

Reilos, 2015-09-07:
same problem on my other samsung tv

Kwiboo, 2015-09-08:
Please try with RasPlex 0.7.1, it contains updated input handling and keymapping.

Reilos, 2015-09-14:
Nope, still the same problem. The (new) user selection screen does not take the numbers on the remote for login credentials.

NedtheNerd, 2015-11-09:
Can you check in latest release and advise if fixed.

Reilos, 2015-11-10:
It's working fine now on my Samsung UE48HU7500. Thanks!

NedtheNerd, 2015-11-10:
Closing issue.

closed 2015-11-10 by NedtheNerd