RasPlex/RasPlex #399: [0.7.0] Incomplete Release Notes

19Gerhard85, 2015-07-20:
Hey guys,
the release notes for 0.7.0 are incomplete.

> If you are using the update function you must add the following lines to your config.txt file, after the update is complete, if you need to enable the HiFiBerry DAC and uncomment the appropriate line for your device, you also need to note the part about lirc-rpi overlay

Which lines should be added? At the moment I have no audio output in playback (HDMI output).

g1t-dlanor, 2015-07-20:
Extra editing of config.txt is only necessary for a few cases:
1: To enable special hardware such as HiFiBerry, which is not an integral part of the RPi
Done simply by uncommenting a line near the end of the config.txt file.

2: To specify the codec license numbers for MPEG2 or VC1 if you bought those licenses for your RPi.
Done by uncommenting the relevant lines in config.txt and replacing dummy null numbers with real ones.

3: To make RasPlex use the composite video output instead of HDMI
Done by entering new HDMI-specific directives into config.txt.
You may need to study forum posts and/or RPi documents to get it working well for you.

4: To make RasPlex adapt the HDMI output for DVI monitors instead of normal TV sets.
Again you may need to study documentation or forum posts to get this right.

For all other cases you do NOT need to edit config.txt in order to make RasPlex work as intended.
Audio output uses HDMI by default, so if you don't get any sound that way something else is wrong.

Note  that unlike most earlier versions the new RasPlex 0.7.0 has a preference setting which allows you to output sound to both the HDMI port and the analog audio port simultaneously. So you should be able to get both those outputs working (without any editing of config.txt).

NedtheNerd, 2015-07-20:
In answer to dlanor's point 1, it's not that simple, if you upgraded the line he is referring to is not there.

I will amend the release notes to contain the missing info, don't know what happened as they were in the original draft.

Regards.

g1t-dlanor, 2015-07-20:
@NedtheNerd: You're absolutely right.
I didn't focus on the 'update' aspect of the text quoted in the original post of this issue.

But what I said still applies without changes to the specific complaint of @19Gerhard85, which was that he didn't get any playback audio through HDMI.
That should have worked without requiring any config.txt editing.

NedtheNerd, 2015-07-20:
Release notes updated.  Closing issue.

closed 2015-07-20 by NedtheNerd