RasPlex/RasPlex #402: RasPlex has no audio when playing AAC 5.1 audio tracks

MarshallOfSound, 2015-07-22:
When I select 5.1 as my desired audio output and then play a movie with AAC 5.1 audio nothing plays through the speakers.  Can we force AAC 5.1 to be transcoded to AC3 5.1 at the server side?

NedtheNerd, 2015-07-22:
AAC5.1 audio plays fine through a suitable receiver - display on receiver shows LPCM 3/2.1.

Check that you have enabled passthrough for all options for capable receiver.

Selecting 5.1 when connected only to a TV is incorrect.

MarshallOfSound, 2015-07-22:
All options are configured correctly (I believe)

An interesting behaviour is when I choose a combination of "Force Transcoding" and "Quality = 20Mbs" it plays 5.1 correctly.  With those two options set to "off" and "Direct Play" respectively it does not

closed 2015-08-03 by MarshallOfSound