RasPlex/RasPlex #403: Audio dual output

yannlet, 2015-07-22:
Doesn't work for me. Anyone else having the same issue ?

g1t-dlanor, 2015-07-22:
Are you sure you configured it correctly ?
Sorry, but I have to ask, since you don't mention how you configured it at all...
In case you missed it you need to set the entry in Preferences > System > 'Audio output' > 'Audio output devices" to the new value "HDMI and Analogue".

yannlet, 2015-07-24:
Yes, I tried exactly this way but doesn't work.
Let me clarify, I need dual output for the following:
I want to hear the sound on my headset (it doesn't have DTS decoder). So I need DTS over HDMI and stereo (analog) through the jack.

NedtheNerd, 2015-07-24:
Dual audio works fine.

In Preferences->System->Audio output:

Set 'Audio ouput' to 'HDMI', this is what is used for video audio output and will output AC3 and/or E-AC3, and/or AAC5.1 and/or DTS to a capable receiver when 'Dolby Digital (AC3) capable receiver' and/or 'Dolby Digital Plus (E-AC3) capable receiver' and/or 'DTS capable receiver' are checked and 'Audio output channels' is set to 5.1.  (If you only output to a TV directly then it will not output sound at all if any are checked and 'Audio output channels' are set to 5.1, for connection direct to TV's this must be set to 2.)

Set 'Audio output device' to 'Analogue', this output is for music audio output only, you cannot change 'Passthrough output device'.

Connect your headphones to the analogue port on the Pi, select any music track and play - sound is output to headphones, no sound is output to any HDMI connected device.

Stop the music playback and select any video media, press play - sound is output to your HDMI connected device (receiver or TV).

I am listening to music on my bluetooth headphones whilst typing this!

NedtheNerd, 2015-07-24:
I've put a post on the Rasplex Forums of how to do this, closing issue.

closed 2015-07-24 by NedtheNerd