RasPlex/RasPlex #404: Choppy video RasPlex 0.7.0

chownsauce, 2015-07-23:
Certain higher bitrate videos I try and watch are very choppy/jerky. The audio is fine it's just the video that is choppy. I've tried toggling different video settings and nothing seems to help. Any suggestions, or is anyone else having this issue?

BradMeier, 2015-07-23:
Please provide mediainfo output for the videos in question, also which version of Plex Media Server and the hardware it runs on, if you are using WiFi or ethernet, RPi1B/+ or RPi2B.

Without this information it is impossible to diagnose the issue.

g1t-dlanor, 2015-07-23:
@chownsauce:
Another factor in this is whether or not your PMS server is using the optional 'secure connections' introduced in recent PMS versions. (uses https instead of http) That will require the RPi to decrypt all incoming data from the server including all media streams. This doesn't load an RPi2 noticeably, but for an RPi1 it lowers the maximum bitrate limit. So that feature needs to be disabled in order to allow RasPlex to play high bitrate videos on an RPi1.

chownsauce, 2015-07-23:
I'm running PMS version 0.9.12.3. The RPi2B and I'm using ethernet. I do have RasPlex running in Berryboot so not sure if that's the issue but I'll test that later. For now I've just downgraded to RasPlex 0.6.0 which runs fine for me in Berryboot. I'm not streaming over https so I'm not really sure what the issue is.

Media
    Video Resolution SD
    Duration 57:07
    Bitrate 1072 kbps
    Width 720
    Height 404
    Aspect Ratio 1.78
    Container MP4
    Video Frame Rate 24p
    Web Optimized Yes
    Has 64bit Offsets 0

Part
    Duration 57:07
    File S03E02 - Bed Bugs and Beyond.mp4
    Size 438.12 MB
    Container MP4
    Has 64bit Offsets 0
    Web Optimized Yes

Video
    Codec H264
    Bitrate 940 kbps
    Bit Depth 8
    CABAC 1
    Chroma Subsampling 4:2:0
    Color Space yuv
    Duration 57:07
    Frame Rate 23.976 fps
    Frame Rate Mode cfr
    Has Scaling Matrix 0
    Height 404
    Level 3.1
    Profile high
    Ref Frames 5
    Scan Type progressive
    Stream Identifier 1
    Width 720

NedtheNerd, 2015-07-28:
Suggest flashing another card with just Rasplex and play same file to see if Berryboot is the issue.

g1t-dlanor, 2015-07-29:
I too have made some tests with Rasplex 0.7.0 with BerryBoot,  and I do get choppy playback with some but not all videos. The affected ones are mainly MP4 files with AVC1 CODEC. Videos with normal AVC CODEC (according to MediaInfo) do not seem to be affected. It might have something to do with kernel differencies, since BerryBoot imposes some common kernel functions of its own on all installed OS.
These tests of mine were made on my RPi2, so RAM or speed issues should not be a factor.

g1t-dlanor, 2015-08-09:
As I suspected the issue was due to BerryBoot using an older kernel than RasPlex 0.7.0, and that problem has now been fixed in the latest release of BerryBoot. You can upgrade to the new version from within the old BerryBoot version, assuming that you run it with an active Internet connection.
This new version can also be downloaded as "berryboot-20150808-pi2-only.zip" from the BerryBoot home site.
Since the problem is fixed I'm now closing this issue. (We should already have done so, as it was not an issue with RasPlex, but with BerryBoot...)

closed 2015-08-09 by g1t-dlanor