RasPlex/RasPlex #407: Wired MS Xbox Controller for PC does not work

Twanislas, 2015-07-28:
Hi !

I'm trying to hook up my wired Microsoft Xbox controller for PC to a Pi 2 running latest Rasplex. Everything works well except this...

Strange thing, the controller's led showing it's "number" is lit up (controller num. 1) so I guess it is correctly detected and driver support is loaded (lsmod shows xpad is loaded) but nothing happens when I press any of the buttons.

I tried booting with the controller plugged in, and by plugging it after bootup.

Can you please have a look ?

Thank you !

H--o-l, 2016-01-24:
Hi !
Same issue here.
If I want to look at the code about that where can I start ?

NedtheNerd, 2016-06-14:
No further comments received, closing issue

closed 2016-06-14 by NedtheNerd

corpeder, 2018-05-24:
i am having this issue and was hoping to get further than the last person the only difference is i am using the wireless usb dongle 

the light shows up and the xpad is detected but no buttons respond

any help would be appreciated. let me know if you need logs, etc...

thejbix, 2019-06-23:
Having the same issue as @corpeder