RasPlex/RasPlex #408: Forced subtitles not selected by default

squawk7600, 2015-08-01:
For containers that hold subtitles (e.g. MKV), they can be designated as forced. Forced subtitles are used for foreign language parts in an otherwise native language movie. This means that, for example, an English spy movie will not generally show subtitles (when shown in England) but when the action moves to Russia and the sneaky Russians are plotting, the subtitles automatically appear. This is what happens in a regular TV movie. In RasPlex however, these forced subtitles are not selected automatically which they should be. One has to go to the menu and switch them on (much like any other subtitle). The "forced" subtitle should be "forced" to appear irrespective of the subtitle selection.

NedtheNerd, 2015-08-01:
Have you configured PMS? In Server, Languages there is 'Subtitle mode', set this to 'Shown with foreign audio', works for me!

squawk7600, 2015-08-01:
Brilliant. Thanks for the tip, I’ll check it out.

squawk7600, 2015-08-04:
I've set the above setting but it has made no difference to playback in RasPlex. The default subtitle selection is still "None" though the forced subtitles are there as a selectable option.

g1t-dlanor, 2015-08-05:
@NedtheNerd: I think I see what @squawk7600 is getting at. It is not a case of needing subtitles because of a foreign audio track, which is what that PMS setting is intended for. It is rater a case of having the correct audio track, but with segments of the movie using yet another language, presumed unknown to all viewers. For such cases video makers have two options. They can either hardcode subtitles onto the video track (thus ruining parts of the pure video footage) or else they can add a forced subtitle track, which should be displayed for all cases except those where a player has a specific setting to suppress it, or where the user has selected a specific normal subtitle track (which should then include all 'forced' subtitle entries in that translation). I agree with @squawk7600 that it is an error not to display such a track when no subtitle has been selected, but I am unsure whether this is a fault of RasPlex or of "Plex Home Theater", whose behavior RasPlex is intended to copy. In the latter case this is really something that Plex Inc coders need to fix.

@squawk7600:
Please test the same videos with similar settings in "Plex Home Theater" on a PC, and report whether this also fails to show the 'forced' subtitles.

NedtheNerd, 2015-08-05:
Totally aware of what you're saying, what I suggested works for me.  After all a forced subtitle track is still a subtitle track, normally srt files.

NedtheNerd, 2015-08-05:
Just to ensure that you understand, my forced subtitles are just that, not a full subtitle track for the whole movie and I never use mkv, I always rip to mp4 and have a separate forced subtitle track, normally obtained from a rip or if I forget, from opensubtitles.  The settings I suggested then work correctly.

Perhaps it's the mkv file that causes the issue.

NedtheNerd, 2015-08-05:
There's also a support page for this now https://support.plex.tv/hc/en-us/articles/204985278

squawk7600, 2015-08-06:
OK after a little bit of trial and error I think I've got my head around it. First thing to say; this is not a RasPlex error as Ned and g1t suspected. This is a Plex omission. The subs appear as per the explanation on the support page given above (noting that the Plex setting "English" works over "English (UK)" which does not, when using MKV merge and choosing "eng (English)" option). 

However, no account is taken of the "Forced" flag at all. So if the MKV audio track is set to "English" (because the movie is predominantly in English with some foreign speaking) and the MKV subs are also set to "English", the subtitles *will* be displayed when Plex is set to "Always Enable" and *will not* be displayed when set to "Show with foreign audio". This is irrespective of the Forced flag - it plays no part. 

The subs only display when Plex is set to "Show with foreign audio" when the audio in the file is different to your preferred setting. Very often, I find MKVs have the audio track set to "und (undetermined)" which means that the subs will show up. Again, nothing to do with the Forced flag.

So I put this here for any subsequent readers coming across this, but to re-iterate, this is not a RasPlex issue - it's PMS. So I will raise it over there at Plex.

NedtheNerd, 2015-08-06:
As the OP has subsequently found that it does work and is neither a Rasplex or PMS issue, issue is closed.

closed 2015-08-06 by NedtheNerd