RasPlex/RasPlex #409: Outdated libpng dependency

NullDivision, 2015-08-03:
It would appear that the .bin file for the latest release is out of date. When trying to execute the BIN file I get the following error:

```
./GetRasplex.bin: error while loading shared libraries: libpng12.so.0: cannot open shared object file: No such file or directory
```

The current stable release for libpng is 1.6. Would it be possible to update the dependency stack to it and have 1.2 as a fallback?

Risca, 2015-09-21:
We use Debian to build the [rasplex-installer](https://github.com/RasPlex/rasplex-installer) and on latest stable Debian Jessie, libpng16 is still in experimental. Once Debian updates it to stable, we will link against it.

closed 2015-09-21 by Risca

NullDivision, 2015-09-22:
So what you're saying is you don't even care about forward compatibility or other distros.

g1t-dlanor, 2015-09-23:
@NullDivision: I'd rather say that it's the developers of your Arch Linux who don't care about backwards compatibility with programs created before their own adoption of libpng16 or about compatibility with other distros where an older lib remains the norm. But in all likelihood there exists some backwards compatibility lib that you can install to eliminate the problem at your end.

Risca, 2015-09-23:
I would say it is practically impossible to dynamically link against both libpng16 and libpng12 at the same time. They are both binary incompatible and have different API, and on top of that, libpng is a an indirect dependency. It is a dependency to the Qt framework, and which version to link against is determined at compile-time of Qt.
The only other alternative is to statically link against a specific version of libpng. You are free to create feature request on the [rasplex-installer repo](https://github.com/RasPlex/rasplex-installer)

NullDivision, 2015-09-24:
Yeah. Sounds a bit tricky. But if the lib is only used as a dependency of Qt, I don't see why it would require libpng12 for the latest version of Qt.

I'll do some reading up and create a request if I have something tangible.

Thank you.