RasPlex/RasPlex #410: No screen refresh after jumping to initial letter

squawk7600, 2015-08-04:
When navigating the list of, say, "All Movies", you can jump down (or up) the list by hitting a letter on the remote. So you can jump straight to the V's or straight back to the A's and so on. This works, but after the jump, the movie box covers are shown as the default "plex-blanks". You have to move sideways to get the tiles to show the box covers again as normal.

NedtheNerd, 2015-08-04:
Which version, what hardware, etc, etc, etc.

squawk7600, 2015-08-04:
Good point, well made....*embarrased*
RasPlex 0.6.0 on an Pi 2 Model B (Quad Core, 1GB RAM)

NedtheNerd, 2015-08-04:
This (and a lot of other important issues) was fixed in 0.7.0, suggest you upgrade.

Regards

squawk7600, 2015-08-04:
Great, will do thanks. Tried searching for the issue before raising but obviously didn't type the right selection of words.

NedtheNerd, 2015-08-04:
No problem.

g1t-dlanor, 2015-08-09:
Since this issue was apparently fixed by RasPlex 0.7.0 I'm closing it now.

closed 2015-08-09 by g1t-dlanor