RasPlex/RasPlex #411: Turtle Beach Audio Advantage Micro II

mihalski, 2015-08-07:
Hi,

I'm trying to get the digital passthrough on the Turtle Beach Audio Advantage Micro II working in RasPlex. Regardless of how I set the "Audio output" it always seems to try and send the audio via HDMI even if I have the ALSA device for the Turtle Beach Audio Advantage Micro II S/PDIF selected for the digital passthrough.

On NedtheNerd's suggestion I tried OpenELEC and with the 5.0.8 release I was able to get Dolby Digital working without a problem simply by setting the output device and digital passthrough to the Turtle Beach Audio Advantage Micro II.

I would be more than happy to donate this device to the RasPlex development community. All I need is an address to send it to.

Thanks in advance.

Regards,
Michal

mihalski, 2015-08-10:
Hi,

I'd like to add that now with purchasing a Pi2 and testing with my TV the optical USB output DOES work for the GUI sounds, just not as a passthrough.

It seems like it's REALLY close to working. Can anyone suggest some tweaking that might be possible from the shell, which MAY help this last piece of the puzzle to be solved? You'd expect digital passthrough to be possible if the GUI sounds are working wouldn't you? It seems like a small leap.

Any help would be greatly appreciated.

Regards,
Michal

NedtheNerd, 2015-08-10:
As I said in the forum this requires a coding change.

mihalski, 2015-08-10:
Are such coding changes in the pipeline at all?

If I set the passthrough to Pi HDMI and try to play something the TV doesn't output any sound, the video gets choppy, the audio output settings say "error: no audio devices" and the Pi freezes.

Still experimenting with different settings to see what is possible and how different audio settings work.

mihalski, 2015-08-10:
In addition to the above if i ssh into Rasplex and download the Dolby Digital WAV file from http://www.mysurround.com/test/TestMySurround-en.dbr.wav.zip  and then play it with the command:

    aplay -D iec958:CARD=MicroII,DEV=0 /storage/music/TestMySurround\(en
\).dbr.wav

It plays correctly. I hope that information is useful.

NedtheNerd, 2015-11-25:
See also #343 #355

NedtheNerd, 2016-06-06:
Can you try this in latest release 1.6.0, thanks.

NedtheNerd, 2016-06-14:
No further update received, closing issue.

closed 2016-06-14 by NedtheNerd