RasPlex/RasPlex #412: Video freezing during direct play

glamorous, 2015-08-15:
Size of Raspberry Pi: Pi2, 512
Type of SD card: SDHC class 10
Internet connection type: Ethernet
Version of RasPlex: latest: 0.7
How did you install rasplex? gui installer OSX

It happens from the first time I have this version (clean install) on almost every file I trying to play after a couple of minutes it started to play. At a sudden moment the image freezes, sounds stops. Sometimes the subtitles go one for a few seconds and then stop to...

I reproduced the error now multiple times and the last times I had the "info"-window open. Some numbers started doing "strange" things when it just happens, they are going fast under zero

P aq
C ad & af

Cache is 0 but I'm asking to cache for at least 20%... CPU are all under the 5% so I guess nothing to do with the CPU power... The file that I'm testing has mp3 sound stereo and 420p video file so not heavy at all.

After waiting for a couple of minutes the number start to rise again (P aq) very slowly.
When I hit rewind and wait a couple of seconds then suddenly it starts rewinding and I can hit play again and it plays the (already) played fragments without problem.

The numbers then:
P (aq 99%, kb/s 2967,72)
C (ad: 5.614 a/v: -3.121 cache: 0 B 100% 0 sec af: 97% vf: 100% amp: 0.00)

Hope you guys are having more info with this. If I can do something else so this problems can be sorted out, I hope... I hadn't very much of succes with all the latest releases :(

Some of my settings (most are default, no overclocking applied):
Use FFMPEG AVIO voor network traffic: ON
Vertical Blank sync: OFF
Use LCD/VFD: OFF
Visualize dirty(?) places: OFF
Cache Size: 20%
Local video quality: Direct Play
Video quality external: Direct Play
Force Transcode: OFF
Transcode always for subs: OFF (apparently subs are working without transcoding on raspberry pi)

NedtheNerd, 2015-09-11:
Can you advise if you still have this issue in 0.7.1, thanks.

dimpan, 2015-09-13:
I'm having the same issue with 0.7.0 and 0.7.1, I can fix this by selecting "always transcode video" but it's only a workaround and my server can't really keep up with the transcode if i start a second stream. 

Using OSMC with the same rPI2 i have no problems with the same videos.

Here's a sample of the movie file ;

https://www.dropbox.com/s/ci531fqm8jgk3fm/sample.avi?dl=0

D(Audio:ac3 [0] [0][0] / 0x2000), 48000Hz, stereo, fltp, 192kb/s)
P(aq:99%, Kb/s:1193.09)
D(Video:mpeg4 (Advanced Simple Profile) XVID / 0x44495658), yuv420p, 720x304 [SAR 1:1 DAR 45:19], 1525 kb/s)
P(fr:23.976 vq: 0%. dc:omx-mpeg4, MB/s:0.66)
C( ad:5.167 a/v:-0.015, edl:-, dcpu:24% acpu 0% vcpi 0% cache:0 B 94% af:89% vf:0% amp: 0.00 )
W( fps:9.55 CPU0:19% CPU1:0.0% CPU2:0.0% CPU3:3.9% ) P( server:servername direct-play )

heres the video info from the .nfo ;

ViDEO: 1600 kbps XviD               

 RESOLUTiON: 720x304                           

 RUNTiME: 1h55m33sec      

 SUBTiTLES: English (.srt)                              

 AUDiO: 192 CBR AC-3 2.0

g1t-dlanor, 2015-09-15:
I've downloaded your sample file and played it through my own PMS server on both my RPi2 and my RPi1 units, and it plays perfectly for both cases. This leads me to believe that your problem is due to some limitation in your PMS setup or its connection to your RPi2, though I know very little about either of those things at your end. Bad WiFi interference is one common reason for stutters in media that would play fine with Ethernet connection (which is what I use). But in this case I'm more inclined to suspect your PMS setup, since you mention that it has trouble keeping up with transcoding when handling any other stream in parallel. If your PMS system is underpowered (CPU-wise), then it may also have problems keeping up with temporary stream bitrate peaks and secure transfer encryption. These problems would never occur when playing the same file through OSMC, as I assume that this involves no Plex server streaming, but just simple SMB file transfer. (Unless you use some PlexBMC addon variant for such playback).

It's hard to pinpoint a definite reason for your problem without more information from you.
Please add more info about your PMS system and its LAN connection to the RPi2.
Also tell us if you made any configuration changes in the RasPlex install.
(like config.txt settings etc.)

dimpan, 2015-09-15:
@g1t-dlanor Thanks for your help! i'm pretty sure the issue isnt performance, everything is hardwired GB ethernet, and i'm able to play 15gb+ mkv 1080p no issue, and osmc has no issue with any file, it's just certain movies when started will chop right from the beginning and never play properly, would it be smarter to host the entire file instead of just the sample?

A friend is having the exact same issue, however his is on wifi and rasplex version 0.7.0, but he's able to play larger files as well, i tried updating to 0.7.1 which is identical.

here's my server info ;
AMD Athlon II X4 640 2.6ghz / 4GB / Win10 x64
Plex Server Version 0.9.12.11

g1t-dlanor, 2015-09-15:
Like I said before, I didn't really think it was the LAN connection that was responsible.
As for the media file sizes, they have no direct influence on any streaming issues. It is the bitrate of the stream that matters, not its size or duration. So a huge file with a very long playtime may have much lower bitrate (and thus less problems) than a smaller file with a very short playtime. Even so the duration of a sample may influence whether or not a problem will show up in testing, as some buffer handling problems build up gradually before a visible problem is triggered.

Since playback in OSMC doesn't involve any Plex streaming, its use proves nothing about which end of the stream is responsible for the errors you get. My personal main suspect remains your PMS system/setup, because that is the main difference between your setup and mine, and I don't get any stutters with your sample file, neither on RPi1 or on RPi2.

If you reliably get the same errors every time you play the short sample you posted, then I should be able to see the same errors here, unless the errors are caused by something other than RasPlex (such as your PMS). But if you do not get the errors with the short sample, then it's of course pointless for me to test with that sample, and I will need the entire file.

Btw: One new feature of recent PMS and the latest RasPlex version is the 'Secure connection' support, which does add some additional CPU load to the stream processing at both ends of the connection, since PMS has to encrypt and RasPlex has to decrypt every byte of every package in the media streams. That CPU load is sufficient to halve the max bitrate of an RPi1, and while it doesn't matter that much to an RPi2, the encryption demands may also be stalling your PMS. I suggest that you test if disabling the 'Secure connections' entry in the PMS settings has any effect on your results with these videos.

Another useful test would be to try playing the same videos in other plex clients than RasPlex, to see if any freezing and choppiness is visible there as well. (Indicating a problem with the streaming of PMS.)

dimpan, 2015-09-15:
I tried the movie on plex home theater, and through the web front end, it played fine, the only time i see an issue is on rasplex. If you want the fuil file is here and i apologize in advance for what it is ;

https://www.dropbox.com/s/tcvl407u1adkntn/Magic%20Mike%20XXL%20%282015%29.avi?dl=0

i disabled secure connection and will try again when i get home, thanks again for the help.

g1t-dlanor, 2015-09-16:
I'm downloading the full file now and will test it later today. As for the tests you made with PHT and Plex/Web clients, I'd like to know if those tests were done on the same computer running PMS, or on some other PC connected to it through the LAN ?

dimpan, 2015-09-16:
two completely separate systems, pms is the same version on both, and both has rpi2's running rasplex, just one was wifi, and one was wired.

i tried disabling secure connections, restarted the plex server and rasplex, and i didnt notice any difference.

g1t-dlanor, 2015-09-16:
I'm not quite sure what you mean by separate systems with PMS on both.
What I wanted to know is whether or not the PHT and Plex/Web clients in your tests were running on the same physical computer as the PMS server delivering the content. Because if they were then those tests did not involve any networked streaming. A client on the same PC as PMS will just directly play the content without any streaming involved. So any problems PMS may have with networked streaming would not appear in such tests.

In the meantime I have now made some tests with the file you posted earlier. I added it to a test section in my PMS server library and then tested its playback with my RPi units. First I used my RPi2 to play the first 10 minutes of the movie, without any freezing or other glitches. I then terminated that playback and switched to the RPi1 unit and started the movie on this, resuming at the point where I broke off the first tests. In this second test I let it play another 10 minutes into the movie, again without any problems of any kind. And of course both these RPi units run the latest public RasPlex release 0.7.1, just like yours. Since you said that your problems with it would appear very quickly after starting playback I see no need to test the full length of the movie (not really to my taste), as any such problems should have appeared several times over in both my 10 minute tests.

The total lack of any problems in my own tests with this movie indicates that your problems are caused by something other than the RasPlex code, since that is identical for your tests and mine.
There has to be some significant difference between our setups that causes your problems, though I have no clear idea what that difference is. It could be some unsuitable RasPlex configuration setting or some problem with the PMS server or the network configuration of the PC it runs on.

dimpan, 2015-09-17:
Hey thanks for all your help again, i really appreciate it, at home i have a PMS running on win10, connected to GB lan, to an RPI2, where i have the issue with just certain films. On my laptop running plex home theater, i don't have that issue, or through a web browser.

At a friends who has a very similar setup to mine, only just through wifi, he has the same issues as me with the same films. Both of our setups are unconfigured, i've basically just installed PMS and added my library, then installed rasplex and logged in, it's a fairly new setup since i'm trying to switch from using kodi/mysql.

i wasn't able to make a video of the issue yesterday as i wasnt home, but if possible is there anything else i can upload, logs, diagnostics to maybe figure out what could be causing this?

g1t-dlanor, 2015-09-18:
All my tests used direct-play for the playback, as checked by the "i" key command in some of the tests.
Most of the tests used my normal fully configured SD cards, but I did flash one SD card especially to test how it would work without any configuration at all. Even then the movie played perfectly here.

Since I can't trigger the problem in any way on either RPi1 or RPi2, I'm fairly convinced that it's not due to any RasPlex coding error. So it has to be due to some issue with your PMS system and/or its configuration/networking. Windows 10 is still very new (I regard it as still experimental), and its use for PMS servers is not at all well documented, so that's a pretty strong suspect right there...
That's also a major difference between our setups, as my PMS system is running under Win7pro_x64 on an Intel i7 CPU.

I can't really think of anything more you can send/post to clarify the issue. Since it isn't registered as any error by either the Plex client or the PMS server, there probably won't be anything of interest in the logs. In a way it would be better if the problem had more serious results, as a proper crash would give us something to debug... But a mere failure to 'keep up' is very hard to analyze.

dimpan, 2015-09-18:
I did a quick test this morning by creating a plex server on my windows 8 laptop, copying the file over and streaming it to my rasplex, and it worked fine, even having the server on wifi.

So the issue seems to be extremely specific

Windows 10 + certain XVID files + rasplex.

Changing any of those 3 elements solves the issue, i have no idea how to proceed.

Kwiboo, 2015-09-18:
@dimpan Do you still have the same issue if you disable FFMPEG AVIO?

NedtheNerd, 2015-09-18:
@dimpan 

Convert the files to MP4 using Handbrake,

dimpan, 2015-09-18:
@Kwiboo 
Thanks that was it!

NedtheNerd, 2015-11-09:
Closing issue, see above comment.

closed 2015-11-09 by NedtheNerd